bancika/diy-layout-creator #702: Suggestion: Right-mouse pan option

mutcode, 2022-04-26:
Hi there!  Loving DIYLC so far.  However I find myself automatically trying to pan the camera with right-mouse click and drag.  As far as I can tell, right mouse button currently serves the same purpose as left mouse.  Would be great to have an option to have this pan the camera around instead.  Many games and editors have this feature, which is probably why it is "muscle memory" for me.  Perhaps also for others, but either way, a useful feature. 

For clarity, I would see it working like this:   
Right mouse click on the background - mouse cursor changes to "hand" icon or something to show that you are now in drag mode.
Move the mouse - camera view moves in the opposite direct, to give the effect of that place you have "grabbed" being "pulled"
Amount of movement of the mouse cursor would exactly match the amount the view is moved by (pixel for pixel).
Example: I click in the top right corner and drag to the middle of the view.  What was in the top right corner is now in the middle of the view.

bancika, 2022-04-29:
Interesting idea, thanks

bancika, 2022-10-04:
Actually, there is already a feature exactly like this but with a middle mouse button, I totally forgot about it. Closing :)

closed 2022-10-04 by bancika