bancika/diy-layout-creator #704: pasting a variant with display name set changes name

EL34xyz, 2022-05-05:
If you save a variant and it is saved to display the name you have entered and then you paste that variant, DIYLC does not display the saved name.
In my example, I saved a new variant electrolytic cap and named it 10uf/500v
When I pasted the variant it pasted the name as C1

DIYLC makes you edit every single pasted variant so it displays the correct name that is saved in the config.xml file

Looking at the config.xml file and finding the 10uf/500v cap
It has saved the name correctly as 10uf/500v
It also has saved the Display as NAME
So the config.xml file is all correct for this variant
`          <org.diylc.core.Template>
            <name>10uf/500v</name>`

The variant has been saved correctly but when you paste a variant it is not using the saved data for this component
`              <entry>
                <string>Display</string>
                <org.diylc.common.Display>NAME</org.diylc.common.Display>
              </entry>`
              
[image]

EL34xyz, 2022-06-28:
Any plans to change this in a future version?

bancika, 2022-07-09:
Variants are doing what they are supposed to, I don't see a bug here per se. The fact that you selected display = Name only means that the name will be rendered and that works. What if the name of the component was C325 when you saved the variant? I doubt that you'd like it to be carried forward to all pasted instances.

I see your point, but it sounds like a feature that would allow the user a choice, not something that should be changed for all cases.

closed 2022-12-14 by bancika