bancika/diy-layout-creator #705: PROBLEMS WITH SAVING TO PNG OR PDF MAKES PRINTING FOR TONER TRANSFER IMPOSSIBLE

baddriddim, 2022-05-08:
Hello. when i save as pdf or png, there's somekind of a grayinsh underlayer that make toner tranfer impossible. I've checked my printer settings and everything is just as usual. i didnt try to use a previous version but i hope i can because most of the circuits i need to print today have been modified with latest version. when i print directly from the software there is no grayish underlayer but printing from the software never worked with me; i get white lines cuting in my circuits.
thank you for reading me and for this great software (although i'm pissedoff right now lol)

baddriddim, 2022-05-08:
ok, i have rechecked everything, so if that can help i went back to version 4.20 (for only using my trace layers to make printing files it worx, but think i remember iread about that sometime) and i saved as a png, well when you see the small image in the windows file you can see one has a perfect white page and the other (made with latest version) is grayish in the background. i also tryed to put a pure white board or saquqre behind but in pdf or png the white color stays grayish no matter what, so i think there must be some kind of mistake within white color configuration, if that can help... i have a few ideas to suggest but i will do that when i can make a substancial donation (i've made small ones a couple times but i make a lot of circuit boards with this software so i think  would be fare from me to make a nice donation when time comes). I saw you needed some building blocks from users, i have a bunch i could send (also i never save as building blocks and make my own files, where should i send those? in the lace where we can share circuits?
thanks for reading me.
BaddRiddim Workshop

bancika, 2022-05-08:
Please attach one of problematic files so I can examine it

bancika, 2022-05-09:
never mind, I was able to reproduce it.

bancika, 2022-06-05:
released in 4.24

closed 2022-06-05 by bancika