bancika/diy-layout-creator #706: DIYLC run.sh file fails to run

cehunter-at-gb-x, 2022-06-06:
Hi

When trying to use the last few versions of DIYLC from a Linux desktop (Mint Linux), the "run.sh" file doesn't run.  The error (when trying to open in a terminal) is shown in the attached image.
[image]

However (despite your warnings) it is possible to run the programme successfully by selecting and running the diylc.jar file.  As far as I'm aware, everything that needs to be set to "allow execution" is enabled.

Can you offer a fix for this?

Regards

Chris

bancika, 2022-06-06:
Hm...the error says that no file is found. Which command do you use?

derchr, 2022-07-04:
The run.sh uses DOS line breaks which causes issues when executing on Linux.
You can use the utility `dos2unix` to convert the file.

Also, as the run.sh is only to be used on Unix systems, it should be changed to use Unix line breaks.

bancika, 2022-07-05:
Damn, that happens all the time because I develop on windows.

bancika, 2022-12-14:
fixed

closed 2022-12-14 by bancika