bancika/diy-layout-creator #707: Selection counter does not list correct amount of items

M0JXD, 2022-06-09:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

Log file:

M0JXD, 2022-06-09:
[log-2022-06-09 06.23.21.txt](https://github.com/bancika/diy-layout-creator/files/8872722/log-2022-06-09.06.23.21.txt)
[image]

bancika, 2022-06-10:
Thanks for reporting, will fix.

closed 2022-06-10 by bancika

M0JXD, 2022-06-10:
Super, thanks!