bancika/diy-layout-creator #709: When moving application to a different screen, the drawing goes black

Spunky9999, 2022-06-17:
I am running 2 monitors and a large format graphics pad. All 3 run the same resolution (1920x1080)
My graphics card is a Nvidia RTX 3060 on Windows 11 Pro 64 bit with ActiveX 12.
When I move the DIYLC program to another monitor either by drag and drop of Windows command Win-Key/Shift-Arrow key, the drawing area of the application turns black. Refreshing and other attempts to change it all fail. The application always opens on Monitor #2. Moving it to extended monitor 1 or 3 causes the drawing area to go completely black. 

This also happened on a Windows XP computer and a Windows 10 computer. Any movement of the application to another screen causes the drawing area to go all black and unusable.

Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the follo
[log-2022-06-17 04.35.15.txt](https://github.com/bancika/diy-layout-creator/files/8931462/log-2022-06-17.04.35.15.txt)
wing:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.
*

bancika, 2022-06-17:
Try playing with hardware acceleration settings in Config

Spunky9999, 2022-06-17:
That fixed it. Thanks

closed 2022-07-31 by bancika