bancika/diy-layout-creator #711: MacBook Pro 16". Just done a complete rebuild of my Mac system etc DITLC was working fine V4.24.0 and java 8 update 241. But now when I open DIYLC it brings up my amp diagram and then if I try and use the % tab tab the bottom it freezes. Need help to fix this.

foggy09, 2022-06-27:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

foggy09, 2022-07-03:
Hi
Sorry I meant by the % it was at the bottom right of the screen under DIYLC window that is where you adjust the size required eg 100%, 50% etc. I brought in the project at 100% then Change to 66% OK but when I went to 33% and then back to 100% the work area went bro blank with nothing showing up.
I have checked that DIYLC is the latest, I have installed Java script java 8 update 241 and checked everything I can.
I also attached the log you required
[log-2022-07-03 10.40.45.txt](https://github.com/bancika/diy-layout-creator/files/9033907/log-2022-07-03.10.40.45.txt)
.

bancika, 2022-07-09:
Does it happen with any project or specific ones?

foggy09, 2022-07-13:
Hi Yes I tested some of my other projects and the problem problem of sizing does not occur.
When I bring up my major project I have been working on for years now it comes up in 100% mode but if I change it to say 33% then the project freezes up
[log-2022-07-13 04.06.02.txt](https://github.com/bancika/diy-layout-creator/files/9099424/log-2022-07-13.04.06.02.txt)
.

bancika, 2022-07-13:
Can you please email me the project file to bancika AT gmail.com..that's
the best way to debug these issues.