bancika/diy-layout-creator #713: netlist

KinyaKanno, 2022-07-03:
How can I import .net file ?
I've tried  "File"-> "Import Netlist" and select .net file created by LTSpice XVII but no components are detected.
What kind of format is available ?

bancika, 2022-07-31:
Do you have a sample file?

lulumahot, 2023-01-19:
I have the same issue in the config:
Mac OS Interl Monterey 12.6
DIYLC 4.3.5.0
I create my netlist with kicad (maybe i'm doind something wrong here, i'm not very used to those softs).
File is below (renamed adding .log ext to join it)
[vco AS3340.net.log](https://github.com/bancika/diy-layout-creator/files/10461549/vco.AS3340.net.log)

greenonline, 2023-06-03:
Using DIYLC 4.37.0

As a test, I created a very simple KiCAD (6.0.11) schematic, and generated a netlist (see below).

When I tried to import it into DIYLC, I got the following dialog:

[image]

and then clicking Ok, not a lot happened. TBH, I am not sure what is meant to happen.

Log file:
```
14:25:16,210  INFO [main] DIYLCStarter - Done hiding splash screen.
14:25:16,267  INFO [SwingWorker-pool-1-thread-3] UpdateChecker - 0 updates found
14:25:16,272  INFO [main] DIYLCStarter - Setting up open file handler for Mac...
14:25:16,274  INFO [main] DIYLCStarter - Injecting default properties.
14:25:16,676 DEBUG [SwingWorker-pool-1-thread-2] PhpFlatProxy - Connecting to: http://www.diy-fever.com/diylc/api/v1/getAnnouncements.php
14:25:16,889  INFO [SwingWorker-pool-1-thread-2] UpdateChecker - Trying to download file: http://diy-fever.com/diylc/api/v1/update.xml
14:25:17,295  INFO [SwingWorker-pool-1-thread-2] UpdateChecker - 0 updates found
14:25:24,969  INFO [AWT-EventQueue-0] ActionFactory - ImportNetlistAction triggered
14:25:36,110  INFO [AWT-EventQueue-0] ConfigurationManager - Saving configuration
14:25:36,166 DEBUG [SwingWorker-pool-1-thread-4] ActionFactory - Importing netlist from /Users/macbook/Documents/KiCAD files/Test_netlist/Test_netlist.net
14:25:51,181  INFO [AWT-EventQueue-0] ConfigurationManager - Saving configuration
14:25:51,237  INFO [AWT-EventQueue-0] Presenter - pasteComponents(org.diylc.clipboard.ComponentTransferable@625b9403, false, false)
14:25:52,750 DEBUG [AWT-EventQueue-0] Presenter - mouseClicked(java.awt.Point[x=1815,y=1458], 1, false, false, false)
14:25:52,752  INFO [AWT-EventQueue-0] Presenter - setNewComponentSlot(null)
14:25:53,584 DEBUG [AWT-EventQueue-0] Presenter - dragStarted(java.awt.Point[x=1893,y=1460], 2)
14:25:53,946 DEBUG [AWT-EventQueue-0] Presenter - dragEnded(java.awt.Point[x=1929,y=1487])
14:25:54,505 DEBUG [AWT-EventQueue-0] Presenter - dragEnded(null)
14:25:58,512  INFO [AWT-EventQueue-0] ActionFactory - ImportNetlistAction triggered
14:26:03,968  INFO [AWT-EventQueue-0] ConfigurationManager - Saving configuration
14:26:04,015 DEBUG [SwingWorker-pool-1-thread-5] ActionFactory - Importing netlist from /Users/macbook/Documents/KiCAD files/Test_netlist/Test_netlist.net
14:33:45,830  INFO [AWT-EventQueue-0] ConfigurationManager - Saving configuration
14:33:45,865  INFO [AWT-EventQueue-0] Presenter - pasteComponents(org.diylc.clipboard.ComponentTransferable@b6c95ec, false, false)
14:33:45,991 DEBUG [AWT-EventQueue-0] DrawingManager - Render stats: {}
14:33:45,991 DEBUG [AWT-EventQueue-0] DrawingManager - Page stats: 19x 13ms avg
14:33:45,996 DEBUG [AWT-EventQueue-0] DrawingCache - Render cache contains 0 elements, approx size is 0 MB
14:33:46,004 DEBUG [AWT-EventQueue-0] DrawingCache - Render cache contains 0 evicted entries
14:33:46,005 DEBUG [AWT-EventQueue-0] DrawingCache - Cache render distribution stats: {}
14:34:38,111 DEBUG [AWT-EventQueue-0] Presenter - mouseClicked(java.awt.Point[x=2723,y=1499], 1, false, false, false)
14:34:38,112  INFO [AWT-EventQueue-0] Presenter - setNewComponentSlot(null)
``` 

Test Netlist file

```
(export (version "E")
  (design
    (source "/Users/macbook/Documents/KiCAD files/Test_netlist/Test_netlist.kicad_sch")
    (date "Saturday, 03 June 2023 at 14:24:52")
    (tool "Eeschema (6.0.11-0)")
    (sheet (number "1") (name "/") (tstamps "/")
      (title_block
        (title)
        (company)
        (rev)
        (date)
        (source "Test_netlist.kicad_sch")
        (comment (number "1") (value ""))
        (comment (number "2") (value ""))
        (comment (number "3") (value ""))
        (comment (number "4") (value ""))
        (comment (number "5") (value ""))
        (comment (number "6") (value ""))
        (comment (number "7") (value ""))
        (comment (number "8") (value ""))
        (comment (number "9") (value "")))))
  (components
    (comp (ref "J1")
      (value "Conn_01x04_Male")
[... 48 lines cut ...]
      (uri "/Applications/KiCad6.0.11/KiCad.app/Contents/SharedSupport/symbols//Connector.kicad_sym"))
    (library (logical "Device")
      (uri "/Applications/KiCad6.0.11/KiCad.app/Contents/SharedSupport/symbols//Device.kicad_sym")))
  (nets
    (net (code "1") (name "GND")
      (node (ref "J1") (pin "3") (pinfunction "Pin_3") (pintype "passive"))
      (node (ref "J1") (pin "4") (pinfunction "Pin_4") (pintype "passive"))
      (node (ref "R1") (pin "2") (pintype "passive")))
    (net (code "2") (name "Net-(J1-Pad2)")
      (node (ref "J1") (pin "2") (pinfunction "Pin_2") (pintype "passive"))
      (node (ref "J2") (pin "3") (pinfunction "Pin_3") (pintype "passive"))
      (node (ref "J2") (pin "4") (pinfunction "Pin_4") (pintype "passive"))
      (node (ref "R1") (pin "1") (pintype "passive")))
    (net (code "3") (name "VCC")
      (node (ref "J1") (pin "1") (pinfunction "Pin_1") (pintype "passive")))
    (net (code "4") (name "unconnected-(J2-Pad1)")
      (node (ref "J2") (pin "1") (pinfunction "Pin_1") (pintype "passive")))
    (net (code "5") (name "unconnected-(J2-Pad2)")
      (node (ref "J2") (pin "2") (pinfunction "Pin_2") (pintype "passive")))))
```
This was the test schematic

[image]

Maybe the schematic was bad, or too simple. I can provide a better test if required.

JustGitting, 2023-11-19:
Hi @bancika, @greenonline,

I've had the same problem with my Kicad schematic I've created and exported as a netlist. The schematic passes the Electrical Rules Checker (ERC) in Kicad so it should be valid. 

I'm new to kicad/diylc so I'm a novice using both tools and may have made mistake somewhere.

Software:
DIYLC V4.37
Kicad V7.0.

@greenonline, Do you have the original KiCad schematic file to check if it's valid?

@bancika,
There seems to be variations in netlist formats across applications (https://electronics.stackexchange.com/questions/1674/is-there-a-common-netlist-format). What file format does DIYLC expect for a valid netlist files?

Another question, how do I export the netlist from a diylc layout directly to a file? 

The only way I found I could do it was by File -> Analyze -> Generate DIYLC netlist (*) which opens a window with the netlist text. 

I then manually copy-paste the text into another text editor and save the file. However, when I try to import the netlist created by diylc back into a new diylc instance, no parts are found. It is the same problem as described previously.
 
I looked for some documentation on this, but could not find anything obvious.

bancika, 2023-11-19:
DIYLC supports LTSpice netlist format at the moment, not KiCad

@JustGitting use "Generate Spice Netlist" feature as DIYLC netlists are my
internal format that is used for some other features under "Analyze" menu

Cheers.

JustGitting, 2023-11-19:
Thanks @bancika for the the clarification.

Is the feature to export Spice netlists to a file on the roadmap?

JustGitting, 2025-02-13:
Hi @bancika, 

I don't want to reopen or create a new ticket, so I'll put this request here.

Do you have a one or more examples of netlist files created by LTSpice that work with diylc import netlist feature?

I've found some example netlist LTSpice files online, but I get the same error. I want to see how it works. Thank you!