bancika/diy-layout-creator #714: 4 band Resistor Orange color

EL34xyz, 2022-07-04:
On 4 band 1/2 watt resistors (4th band is tolerance and not shown)
Can you make the Orange band Orangier?
The Orange is so close to Yellow that unless I display the value, it's hard for me to tell if the 3rd band is Orange or Yellow?

It's easier to tell if you put a 10k right next to a 100k
Then you can see a slight color difference
But a stand alone 10k can get mistaken for 100k because it looks closer to yellow than a deep orange

[image]

bancika, 2022-07-04:
Okay, that's easy enough...

[image]

closed 2022-07-04 by bancika

EL34xyz, 2022-07-04:
Thanks Bane