bancika/diy-layout-creator #715: DPDT (Center On) switch (On/On/On) request

PenBach, 2022-07-06:
Any chance we could get a DPDT (Center On) switch in DIYLC?

I'm sure you know what they are, but just to make sure we're talking about the same thing:
They are also called DPDT On/On/On.
Preferably "Type 1", but obviously if you could also add "Type 2" at the same time, that would be fantastic.
When viewed from below
"Type 1" - when in the center position, the middle/common terminals connect to the top left and bottom right terminals.
"Type 2" - when in the center position, the middle/common terminals connect to the bottom left and top right terminals.
Thanks.

bancika, 2022-08-27:
Done for 4.26

closed 2022-08-27 by bancika

PenBach, 2022-10-04:
Thank you so much for this. Superb fast response. Much appreciated.