bancika/diy-layout-creator #716: Music Man Stingray Bass Humbucker Pickup request

PenBach, 2022-07-13:
As the title says, I'd like to request the addition of a Music Man Stingray Bass Humbucker Pickup to DIYLC.
I'm currently using two P bass pickups as a substitute, but it would be nice to have a Stingray pickup.
Thanks.

bancika, 2022-10-04:
Released in 4.29

closed 2022-10-04 by bancika

PenBach, 2022-10-04:
Thank you so much for this. Superb fast response.
There is one thing though - it's a lot bigger than it needs to be (IMHO) and takes up a lot of space when drawing a layout.
All guitar pickups seem quite big to me.
I can't see a way of reducing the size of pickups, but I am quite new to DIYLC.
Any chance of being able to size objects - especially pickups - if that's not possible at the moment?

bancika, 2022-10-04:
They are true to size. You can always un-zoom to fit more on the screen