bancika/diy-layout-creator #718: Guitar Wiring Component Request

Hasselhoof, 2022-07-24:
Hi

Looking for more guitar wiring components for wiring diagrams.

More specifically
DPDT(ON/ON/ON)
Stacked single coil (3 wire)
Schaller Mega Switch
Freeway Switch
Rotary switches
Switched pots

Thanks in advance if anyone can help.

BenWaSoul, 2022-08-28:
Hi. 

Good requests. And looking for QPDT (On/On/On) Type 2.

You can actually set a Single Coil as Humbucker in polarity option and link the 2 finishes to add to it a 3rd wire.

bancika, 2022-08-28:
DPDT(ON/ON/ON) added in 4.26
For switched pots you can combine regular pot with a switch...that's the preferred way as you have flexibility of configuring the pot and the switch individually. You can save them together as building block if you need a one-click-solution.

InZillaWeTrust, 2022-09-29:
I would second the Schaller Mega Switch or even just a generic Import Style 5 Way switch.

bancika, 2022-09-29:
5 way switch is already there under Lever Switch. I'm working on Schaller
Mega.

bancika, 2022-10-04:
Schaller Megaswitch released in 4.29

bancika, 2022-10-07:
Adding 4-pole LP switch here, closing #701

bancika, 2022-10-23:
Rotary switches added in 4.30

StateOfEpicicity, 2023-02-18:
I'd like to request dual concentric as well as dual gang pots.  DIY Layout Creator is amazing.

bancika, 2023-02-18:
You can just use two potentiometer components stacked together for the same
effect. And also you get more flexibility to place them any way you want

StateOfEpicicity, 2023-02-18:
No problem; that's what I've been doing.  Thanks for responding, and thanks so much for your program; it's absolutely the best tool for brainstorming this kind of thing.

bancika, 2023-02-18:
Glad to help!

bancika, 2025-02-11:
remaining: Freeway switch

StateOfEpicicity, 2025-02-11:
The only breakdown I've found of the switching logic outside of trying to decipher freeway's diagrams is here, if you click on the spoilers:

https://guitarnuts2.proboards.com/thread/9752/blade-switches-truth-tables-photos

This will save a load of time if you tackle this!

bancika, 2025-02-11:
The biggest issue with the freeway switch is to render it properly. It's
tricky to draw

StateOfEpicicity, 2025-02-11:
Hmmm, I wonder if you simplified the images of them, just really focusing on their connectors only.

bancika, 2025-02-12:
[image]

made some good progress

StateOfEpicicity, 2025-02-13:
Looking great!  Funnily enough, when you zoom in on their pdfs, their drawings look jaggedly rendered.  Even though this is a screen shot, it actually looks cleaner!

bancika, 2025-02-13:
Thanks. This is a screenshot of my app but I had to draw each piece
manually. As you said, the PDFs are not usable, so I had to trace them and
draw my own version.

bancika, 2025-12-29:
all complete, closing

closed 2025-12-29 by bancika

mm-rnd, 2026-01-04:
Late to the party, but I've been using the Freeway Ultra 3x3-05 switch so have been working on adding it. This is what it looks like and the icon. @bancika  what do you think?

[image]

[image]

mm-rnd, 2026-01-04:
Yeah the blade one is basically specifically for a 3 pickup S type guitar to give more options. They're just not designed as a universal switch. But I had a look over the example configurations for the 3 pickup blade and it's actually not too bad to figure out. The pad labels are a good hint.

bancika, 2026-01-06:
> @bancika <https://github.com/bancika> merged the 3x3-05 without even a
> comment on the pull request. Are you planning on doing the Freeway blade
> switches @bancika <https://github.com/bancika> or else I might take a
> look sometime.

hey hey, was I not supposed to merge? You can use draft PRs if still not
ready for merging.
I didn't plan on doing blade soon.

Cheers

bancika, 2026-01-06:
> @bancika <https://github.com/bancika> of course you can merge if you were
> happy with it.
> I had asked on the PR if there was anything I didn't do, but you did the
> extra house keeping things straight after.
> I might take a look at the Freeway blade, just to figure out the switching
> logic as a brain teaser.

I appreciate your effort!

Branislav Stojkovic
www.diy-fever.com