bancika/diy-layout-creator #719: DIYLC causes system out of memory (100+GB use) and force quit

nathantlong, 2022-07-28:
After a bit, probably 45 minutes-1 hour, either actively working on a layout or sitting idle, DIYLC is using so much system memory that MacOS says system is out of application memory and causes force quit.  Better have saved your work often!  

[log-2022-07-28 09.26.25.txt](https://github.com/bancika/diy-layout-creator/files/9210104/log-2022-07-28.09.26.25.txt)
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

bancika, 2022-07-28:
What does the memory bar (bottom right corner) show when that happens? It
shows memory usage when you hover over it.

Do you have any specific projects this happens with or it's always?

Also, the log you attached is not the right one, it only has info that
covers two minutes of the app usage while nothing is done.

bancika, 2022-11-28:
Hi, is this still an issue?

bancika, 2025-02-13:
Closing due to inactivity

closed 2025-02-13 by bancika