bancika/diy-layout-creator #721: run.sh only works if run from the app directory

SpinningVinyl, 2022-08-22:
Hi, the current run.sh script works only if started from the same directory the app is installed to.

For example, in my case it's /home/pavel/Apps/DIYLC/my home directory

If I try to run it from any other directory, e.g. my home folder, it gives a classpath error, because the script uses relative paths.

I came up with the following version of run.sh that works from anywhere:

```
#!/usr/bin/env bash
JAVA_BIN="$(which java)"
DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )"
JAVA_AGENT="${DIR}/lib/jar-loader.jar"
CLASS_PATH="${DIR}/diylc.jar:lib"
exec "$JAVA_BIN" -Xms512m -Xmx2048m -javaagent:"$JAVA_AGENT" -Dorg.diylc.scriptRun=true -Dfile.encoding=UTF-8 -classpath "$CLASS_PATH"  --add-opens java.base/java.util=ALL-UNNAMED --add-opens java.base/java.lang=ALL-UNNAMED --add-opens java.base/java.text=ALL-UNNAMED --add-opens java.desktop/java.awt=ALL-UNNAMED --add-opens java.desktop/java.awt.font=ALL-UNNAMED --add-opens java.desktop/java.awt.geom=ALL-UNNAMED  org.diylc.DIYLCStarter "$@"
```

I tested it in Linux and it runs fine, however I have not tested it in macOS.

timsavage, 2022-10-23:
There is a little more to this. Setting a different working directory also causes language translation to fail to load.

M0JXD, 2024-04-06:
I saw in a separate issue that some people have been adding a cd at the top of the script to fix this.

Perhaps that could be automated like so:

bancika, 2025-03-11:
Included for 5.1.0

closed 2025-03-11 by bancika