bancika/diy-layout-creator #726: Mac won't open 4.28

MarsManiac, 2022-09-28:
Hi, 
I downloaded DIYLC 4.28 successfully on MacOS 12.6.  It said 'couldn't verify developer.  Wouldn't let me open it using the usual Mac 'Are you sure?' mechanism.  Went back to 4.27 and it downloaded and opened as per normal.  This update is low priority for me so no worries, just wanted you to be aware.

Be well and thanks!

MarsManiac, 2022-10-11:
Yes, I tried those steps.  That is the usual routine I have to go through to open a new version of DIYLC: go to Finder, right-click, and Open.

But this time, for version 4.28, the dropdown menu did not give me the option to open the app anyway after the warning appeared.  I had no way to override the behavior like I usually do.

Like I said, this update isn’t crucial for me, but if you have anything else you want me to try, I’m glad to help.

Thanks

Hi,
I just tried it on a mac 11.6.5 and it worked okay. Did you try these
steps?
https://support.apple.com/guide/mac-help/open-a-mac-app-from-an-unidentified-developer-mh40616/mac

bancika, 2022-11-28:
Hi. Does the latest work?

MarsManiac, 2022-11-28:
Yes.  I picked it up again about a week later and that version came up perfectly.  Thanks!

closed 2022-11-29 by bancika