bancika/diy-layout-creator #727: Can we change th size of objects, e.g. guitar/bass pickups ?

PenBach, 2022-10-04:
Can we change th size of objects, e.g. guitar/bass pickups ?

They currently seem to take up a lot of real-estate on a layout and I can't see a way of resizing them.

bancika, 2022-10-04:
One of the postulates of DIYLC is that everything is true to size, so you
can plan your layouts accordingly. You can only zoom in/out

bancika, 2022-10-05:
For printing you can typically select paper size so it will scale. PNG/PDF are true to size, but that shouldn't be a problem, as we are exporting as much information (pixels) as possible. For display it doesn't matter because you can zoom in/out and if that file needs to be printed, it can be scaled at that point.

closed 2022-10-22 by bancika

PenBach, 2023-03-14:
> One of the postulates of DIYLC is that everything is true to size, so you can plan your layouts accordingly.

While I appreciate that, could it not be that DIYLC works exactly as it does now, where "everything is true to size", but if some heretic wanted to reduce the size of a component to make the diagram easier to read, then it would be nice to have that option.
I know DIYLC does not currently have any speakers, but if for example you added a 4 x 12" speaker cabinet, would that be absolutely massive?