bancika/diy-layout-creator #728: Saved Files will not open ..

Fankeeee, 2022-10-04:
I keep getting  the following  when I attempt to open previously saved files

"Cannot Launch from folder that contains a "l" in its name "

I have to first open the Application proper and then open whatever file I want under "File"-"Open"
Which allows me to use as normal 

Mac 10:11:16   El Capitan  Configured to open all 3rd party 
thx for any thoughts ..

Fankeeee, 2022-10-04:
[image]

bancika, 2022-10-04:
How exactly does it happen? You fouble click on a file in finder or open it
through application's file/open menu?

Fankeeee, 2022-10-04:
My working file is on desktop .. I double click and the above
shows up.. Previously never was an issue .. Same think if I 
try to open from backup folder .

Fankeeee, 2022-10-04:
Just attempted to open from my lap top versus desktop and files open with
no issue  Maybe I have a hack ??

Fankeeee, 2022-10-04:
Works on my laptop but not on my desktop. i’ve downloaded latest app but to no avail. The banner mentions “folder” which doesn’t exist on desktop

bancika, 2022-12-26:
been trying to figure out what's going on with no success. I suspect it has to do with the mac version as I have no better clues, is the same OS on both of your machines?

Can you please open the Console after it happens and see if there's any more clues there?

[image]