bancika/diy-layout-creator #730: slow on M1 mac

vertigocycles, 2022-10-12:
I'm running macOS 12.6 on an M1 mac with 64G of ram.  DIYLC has been running terribly slowly on this new mac.  I looked under System Report - Software - Applications and it looks like DIYLC is running in Intel emulation mode.  This is the latest 4.29 version (and the previous three that I have installed)  Is there a version that works on M1 natively?

bancika, 2022-10-12:
Hey, can you try to run this build? I don't have an M1 machine to try with, but it should work.

http://www.diy-fever.com/temp/diylc-4.29.0-osx-m1.zip

vertigocycles, 2022-10-12:
Thank you for the lightning fast response! Unfortunately, it has an error "unable to load Java Runtime Environment"

Sean

Sent with [Proton Mail](https://proton.me/) secure email.

bancika, 2022-10-12:
Okay, found another config to play with, can you please try this
http://www.diy-fever.com/temp/diylc-4.30.0-osx-m1.zip
<http://www.diy-fever.com/temp/diylc-4.29.0-osx-m1.zip>

vertigocycles, 2022-10-12:
This time I'm getting, "The application "DIYLC" can't be opened."

Sent with [Proton Mail](https://proton.me/) secure email.

bancika, 2022-10-12:
did you try this?
https://www.asavvyweb.com/macbook/how-to-fix-cant-open-apps-or-install-apps-on-m1-macbook-pro-or-air-macos-monterey

vertigocycles, 2022-10-12:
I have.  I have always setup my computers to allow this.  The result usually is that the first time I try to open (right click, open) it prompts me to send it to the trash bin.  The second time, I'm able to open the application, and every time after that, I can just double click the icon.

bancika, 2022-10-14:
As a workaround you can use the universal package that has no JRE supplied and run it with your JRE built for M1. In the meantime I'll try somehow to make the app bundle work with a M1 JRE.

bancika, 2022-10-22:
hi @vertigocycles can you please try this build? http://www.diy-fever.com/temp/diylc-4.30.0-osx-m1.zip
I tried another config change, no guarantees :/

vertigocycles, 2022-10-24:
> hi ***@***.***(https://github.com/vertigocycles) can you please try this build? http://www.diy-fever.com/temp/diylc-4.30.0-osx-m1.zip

Thank you. I’m away from home for a while but I’ll check it out when I return and will report back

vertigocycles, 2022-11-01:
Just tried it...no dice.  Another "cannot be opened" error.  I opened the package and tried to run the applauncher on its own and received a "cannot load Java Runtime Environment" error.  

After a quick search, it looks like Oracle doesn't yet have a JRE for the M1 chip.  I'm far from being an expert on this, but I'll try to find a workaround with a non-oracle build and will see how it goes.

bancika, 2022-11-01:
I used Azure Zulu JRE bundled with those builds which is supposed to work
on M1

vertigocycles, 2022-11-10:
I've installed the Azure JRE v19 on it's own.  Should I be able to open the DIYLC.jar file and expect it to run?

bancika, 2022-11-10:
Yes, you can use run.sh

vertigocycles, 2022-11-10:
apologies, we're getting out of my limited knowledge on how to operate my computer.

if I open Terminal and type "sh" and then the path to the diylc.jar file, it should run?

bancika, 2022-11-10:
not sure, but you should be able to run with ./run.sh or sh run.sh

vertigocycles, 2022-11-10:
is diylc.jar the correct file to run?

vertigocycles, 2022-11-10:
Assuming that's the correct jar file to run, this is the current error:

/Applications/DIYLC4.30j17.app/Contents/Java/diylc.jar: line 1: PK: command not found
/Applications/DIYLC4.30j17.app/Contents/Java/diylc.jar: line 2:M=WU: command not found
/Applications/DIYLC4.30j17.app/Contents/Java/diylc.jar: line 32: syntax error near unexpected token `)'
/Applications/DIYLC4.30j17.app/Contents/Java/diylc.jar: line 32:L=WUN?NT`org/diylc/DIYLCSplash$1.class}S?n?@=?\ܸNҖ?$?R?8w???)AE???*?ֲ#?)坯??s$.H|O|B?ٴR?h?????????￿|pc?㢊r*?X??4?'

bancika, 2022-11-11:
No, you need to start the run.sh, not the jar directly.

vertigocycles, 2022-11-11:
run.sh doesn't appear in any of the directories underneath the app container

bancika, 2022-11-11:
Oh you are using app container. For this method you need the universal build

bancika, 2022-12-17:
Hey, I found something new, supposedly M1 Macs have issue with unsigned/unofficial apps like DIYLC. Can you try to self-sign it before running? The latest m1 build is  http://www.diy-fever.com/temp/diylc-4.35.0-osx-m1.zip

`sudo codesign --force --deep -s - DIYLC.app`

vertigocycles, 2022-12-18:
I self-signed the app and tried to open it and still get the same error: "The application DIYLC can't be opened"

I tried to open it from terminal using the "open" command and this is the error it returned: The application cannot be opened for an unexpected reason, error=Error Domain=RBSRequestErrorDomain Code=5 "Launch failed." UserInfo={NSLocalizedFailureReason=Launch failed., NSUnderlyingError=0x60000081a550 {Error Domain=NSPOSIXErrorDomain Code=111 "Unknown error: 111" UserInfo={NSLocalizedDescription=Launchd job spawn failed}}}

Sent with [Proton Mail](https://proton.me/) secure email.

bancika, 2022-12-18:
Thanks. Let's try one more thing before running, maybe the zulu JRE is
causing permission issues.

chmod -R 755 DIYLC.app

vertigocycles, 2022-12-18:
tried it and this is what it returned:

usage: chmod [-fhv] [-R [-H | -L | -P]] [-a | +a | =a [i][# [ n]]] mode|entry file ...
chmod [-fhv] [-R [-H | -L | -P]] [-E | -C | -N | -i | -I] file ...

Sent with [Proton Mail](https://proton.me/) secure email.

vertigocycles, 2022-12-19:
I realize I'm just trying random things here but I "opened contents" of the latest application and navigated to "MacOS\JavaAppLauncher" and attempted to run it. I get an error window that pops up with "Unable to load Java Runtime Environment" and my terminal window shows this:

2022-12-18 16:00:23.267 JavaAppLauncher[97271:5965178] Error launching JVM Runtime (zulu-8.jre) Relative Path: '/Applications/DIYLC4.35M1.app/Contents/PlugIns/zulu-8.jre' (dylib: /Applications/DIYLC4.35M1.app/Contents/PlugIns/zulu-8.jre/Contents/Home/lib/jli/libjli.dylib)
error: Unable to load Java Runtime Environment.

Sent with [Proton Mail](https://proton.me/) secure email.

vertigocycles, 2022-12-19:
one more thing:

I tried to open diylc.jar on a generic version and after getting a warning message about functionality, DIYLC will open, but I can't navigate through my folders or select any of my projects.

Sent with [Proton Mail](https://proton.me/) secure email.

bancika, 2022-12-19:
Did you try to run the generic version using run.sh? The terminal command is

./run.sh

Thank for your help btw, I have no way of trying M1 myself.

jcxplorer, 2022-12-27:
I can confirm that the generic version works fine on an M2 MacBook Air.

```
❯ java -version
openjdk version "11.0.16" 2022-07-19 LTS
OpenJDK Runtime Environment Zulu11.58+15-CA (build 11.0.16+8-LTS)
OpenJDK 64-Bit Server VM Zulu11.58+15-CA (build 11.0.16+8-LTS, mixed mode)
```

bancika, 2022-12-27:
Great, thanks! Just need to make the bundled app work too :)

DeclanMorrison, 2023-07-03:
Sorry to resurrect this old issue, but I'm having a similar issue. I've downloaded the OSX version of the latest build (v4.37.0) and I'm also seeing some performance issues, as well as the app running in Intel compatibility mode. I did try downloading the universal build for v4.37.0 and running with my own JRE via `./run.sh` but it gives the following errors:

`
11:37:22,093 ERROR [main] DIYLCStarter - Major error while starting DIYLC
java.lang.IllegalAccessError: class org.diylc.utils.ResourceLoader (in unnamed module @0x1de5f259) cannot access class com.apple.eio.FileManager (in module java.desktop) because module java.desktop does not export com.apple.eio to unnamed module @0x1de5f259
	at org.diylc.utils.ResourceLoader.getFile(Unknown Source)
	at org.diylc.lang.LangUtil.load(Unknown Source)
	at org.diylc.lang.LangUtil.configure(Unknown Source)
	at org.diylc.DIYLCStarter.runDIYLC(Unknown Source)
	at org.diylc.DIYLCStarter.main(Unknown Source)
Major error while starting DIYLC
java.lang.IllegalAccessError: class org.diylc.utils.ResourceLoader (in unnamed module @0x1de5f259) cannot access class com.apple.eio.FileManager (in module java.desktop) because module java.desktop does not export com.apple.eio to unnamed module @0x1de5f259
	at org.diylc.utils.ResourceLoader.getFile(Unknown Source)
	at org.diylc.lang.LangUtil.load(Unknown Source)
	at org.diylc.lang.LangUtil.configure(Unknown Source)
	at org.diylc.DIYLCStarter.runDIYLC(Unknown Source)
	at org.diylc.DIYLCStarter.main(Unknown Source)
`

My Java environment looks like this:
`
openjdk version "17.0.7" 2023-04-18 LTS
OpenJDK Runtime Environment Zulu17.42+19-CA (build 17.0.7+7-LTS)
OpenJDK 64-Bit Server VM Zulu17.42+19-CA (build 17.0.7+7-LTS, mixed mode, sharing)
`

Am I using the wrong Java environment or otherwise missing something earlier in the thread? Please let me know if you need any additional information, and thanks again for making this app, it's really amazing!!

bancika, 2023-07-03:
Can you please try to replace the contents of run.sh with the following and
try again

#!/bin/bash
exec java -Xms512m -Xmx2048m -javaagent:lib/jar-loader.jar
-Dorg.diylc.scriptRun=true -Dfile.encoding=UTF-8 -cp diylc.jar:lib
 --add-opens java.base/java.util=ALL-UNNAMED --add-opens
java.base/java.lang=ALL-UNNAMED --add-opens java.base/java.text=ALL-UNNAMED
--add-opens java.desktop/java.awt=ALL-UNNAMED --add-opens
java.desktop/java.awt.font=ALL-UNNAMED --add-opens
java.desktop/java.awt.geom=ALL-UNNAMED --add-opens
java.desktop/com.apple.eio=ALL-UNNAMED org.diylc.DIYLCStarter "$@"

DeclanMorrison, 2023-07-03:
Sure thing. I actually ended up needing to add the following lines to `run.sh` to get the app to fully launch:
```
--add-opens java.desktop/com.apple.eio=ALL-UNNAMED
--add-opens java.desktop/com.apple.eawt.event=ALL-UNNAMED

bancika, 2023-07-03:
Awesome, thanks! It works now? I'll update the official build

DeclanMorrison, 2023-07-05:
No problem! There was also an additional parameter I added `--add-opens java.desktop/com.apple.eawt=ALL-UNNAMED` which cleared up a non-fatal error. Also, I was mistaken that it was running in Intel compatibility mode, when run directly with `run.sh` the entry in Activity Monitor is called "DIYLCStarter", which is indeed running in Apple Silicon mode!

bancika, 2025-02-09:
Fixed with 4.39

closed 2025-02-09 by bancika