bancika/diy-layout-creator #734: Snap to problem

BenWaSoul, 2022-10-24:
I encounter some issues with the fonction "Snap to" (none / grid / component). No snap at all with the "sticky points" active. I have changed the snap to component to work with the Import and Megaswitch but nothing snap on since. By the way, it seems to be impossible to reactivate it to grid, so I lost the super usefull snap...

bancika, 2022-10-24:
"sticky points" and "snap to" features are related but in different situations. Sticky applies only to moving components that are already snapped to other components.

I see one issue with snap to components that only works when you move existing components around, not when creating new ones. Then there's no snapping to components, but snapping to the grid works in all circumstances.

what do you mean by "impossible to reactivate"? Did you toggle it back on?

BenWaSoul, 2022-10-24:
Thank you for you reply. I mean I select Component, no snap. Then reactive Grid, no snap ?? The Grid option stays unselected. I must re-install DYLC to retrieve the "Snap to Grid" function ON. So, It's okay now but may it be an issue ?

bancika, 2022-10-24:
I don't understand, why don't you just change Snap To back to Grid in
Config menu?

BenWaSoul, 2022-10-24:
Changing snap from Component to Grid didn't work. I select Grid, then reopen the Config menu but Grid isn't selected. So no more snap to grid, after changing it to component.

bancika, 2022-10-24:
can you please send me few of the latest logs? Something must be preventing
it from saving the change in the config.
You can find them in Help -> User Files then go to logs folder. Or just
navigate to ~/diylc/logs.

BenWaSoul, 2022-10-24:
[log-2022-10-24 10.35.12.txt](https://github.com/bancika/diy-layout-creator/files/9850422/log-2022-10-24.10.35.12.txt)

BenWaSoul, 2022-10-24:
The issue comes from the French language, it actually works perfectly in English.

bancika, 2022-10-24:
Ha, it seems to be the issue with translation. Thanks will fix.

closed 2022-11-01 by bancika