bancika/diy-layout-creator #735: Ground shematic symbol doesn't work anymore

BenWaSoul, 2022-10-29:
Hello. Not a real issue but I note using the "Analyze guitar diagram" tool that the Ground shematic symbol doesn't work anymore with any of my existing wiring with which it worked before, nor the new one. Is there something to do to correct that ?

bancika, 2022-10-29:
Hm, did it ever work? In the next release I plan to make it work but it
doesn't look like it ever worked intentionally

BenWaSoul, 2022-10-29:
I always checked my diagrams before making a png of them in my library, and they worked for the most part by adding the symbol on the Jack. Now they don't.

bancika, 2022-10-29:
Can you please send an example? I never use it myself

BenWaSoul, 2022-10-29:
I think this one worked :
[image]

bancika, 2022-10-29:
I see, this was an unintentional bug introduced in the last
version...already have the fix that will go out soon

closed 2022-11-01 by bancika