bancika/diy-layout-creator #736: Analyze Guitar Diagrams doesn't apprear to work.  The Net List does.  I'm a new user and maybe I am doing something wrong?  Also can't find a Push/Pull Pot.  Ubuntu 22.04 user.

jreising326, 2022-10-30:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

bancika, 2022-10-30:
Did you read the instructions?
Can you upload a diy file that doesn't work? You might need to change the
extension to txt.
As for push pull, you can use a regular pot plus a toggle switch. Works and
looks like it.

Going forward please keep one issue per ticket. Makes tracking easier

BenWaSoul, 2022-10-30:
Hi. If your are using the Ground schematic symbol, the analyzer tool doesn't read it actually. Otherwise it works perfectly. Check all the connection with the Snap to Grid engaged and make sure the Jack is wired correctly as the tool needs a complete wiring to analyze.

BenWaSoul, 2022-10-30:
Exemple of a working diagram with DPDT MiniToggle switches close to the pots as Push/Pull :

[image]

jreising326, 2022-10-30:
I am using Snap to Grid, the GND symbols and I downloaded several different *.diy's including the author's Travelcaster example posted in several forums and don't get the same type of report shown in the Forums.  I get a similar "No Pickups are Detected" result in the Analyze Guitar Wiring report.  Running 4.30.1 installed via Flatpak on Ubuntu 22.04.  No .diylc directory in my home folder so I couldn't find the logs.  Also could not build from source code.

bancika, 2022-10-30:
There's a new bug with gnd component, that might be an issue. You can use
version 4.29 while I release 4.31 with the fix. Alternatively, just use
wires to connect all grounding points together.

bancika, 2022-10-31:
Please try v4.31 that just got out

jreising326, 2022-10-31:
I could not find it in Flatpak yet, but it does run under Wine (sort
of, the UI is a little "glitchy" in wine).  However the Analyze Guitar
Wiring is working correctly!  

Very impressive!  Thanks!

jreising326, 2022-11-01:
So Flatpak was updated and installed V4.31, however the Guitar Wiring Analysis still seems buggy.  I'll open a new ticket.

closed 2022-11-01 by bancika