bancika/diy-layout-creator #737: Analyze Guitar Wiring Issue

jreising326, 2022-11-01:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

jreising326, 2022-11-01:
Analyze Guitar Wiring still appears to have a bug.  There could be an error in my own diagram, so I loaded the Travelcaster.diy and ran the Analysis and it didn't generate a full report.  "No pickups are detected in this switch configuration".  On my own diagram, using a Superswitch with GND connections for Coil Splits, the Neck Pickup is Coil split in all connections other than position #1.  I removed all GND's from the Superswitch, which should prevent coil splitting but it still thinks there are some coil splits.  Are the log files in a different location, I don't have a .diylc directory anywhere in my home directory.  Running Ubuntu 22.04 and diylc V4.31 just released tonight.

jreising326, 2022-11-01:
here's some screenshots.
[image]
[image]
[image]
[image]

bancika, 2022-11-01:
Hi!

As for the telecaster diagram, it works as expected. In that drawing none
of the pickup outputs are actually connected to anything because it's just
one thick wire that looks like dimarzio wire. This was drawn before the
wiring analyzer existed.

As for the other one, please upload your DIY file, that is the only way for
me to run the file through the debugger. You might need to change the
extension to TXT.

jreising326, 2022-11-01:
You can close the ticket.  The issue with my DIY was a wiring error.  Your analyzer actually was right the first time around.  I built a manual Netlist for each position to confirm this.  Please close the ticket and thanks again!  This feature alone is super helpful when dealing with Superswitch Wiring w/ some advanced mods, which is my next project.  The Firenza one was just my practice run on something I thought I already had right.  I did not.  :)

bancika, 2022-11-01:
Glad you found it useful!

closed 2022-11-01 by bancika