bancika/diy-layout-creator #739: Analyze Guitar Wiring Bug?

jreising326, 2022-11-05:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

jreising326, 2022-11-05:
I think I found a bug.  Analyze Guitar Wiring is only reporting 4 of the 5 switch connections.  I'm attaching screenshots of the Report as well as the Netlist.  I also reversed the Bridge and Neck Pickup by swapping the Black and Red wires at the top, and then the Wiring Analysis reports correctly.  I guess maybe there could be an issue in my diagram so also attaching the diy as a txt.
[image]
[image]
[firenza3.txt](https://github.com/bancika/diy-layout-creator/files/9944630/firenza3.txt)

bancika, 2022-11-06:
The report is claiming that positions 4 and 5 are identical, that's why they are grouped together.

It's impossible for me to tell if it's right or wrong without spending a lot of time on analysis which may easily turn out to be a wiring mistake.

I suggest trying the following:

* give it a second look, in case it's a trivial mistake
* try to simplify the wiring scheme to the minimum while still keeping the problematic behavior (maybe remove switches, pots, caps, other pickup), something I can work with

jreising326, 2022-11-06:
I did what you suggested and found the problem.  Thanks for pointing out positions 4 and 5 were identical, I did not catch that.  My hand written Netlist was off and a connection for position 4 was missing.  What convinced me it was a bug was I had swapped the Neck and Bridge wires, and with the Bridge in the Neck position it "worked" as expected and showed all 5 of the expected connections.  The source of that diagram is a hand written diagram posted on the Peavey Forum, and I had already found 1 error in the diagram thanks to your tool.  I've now posted the corrected diagram there for reference so others don't end up rewiring their guitars as many times as I have done mine. :)  Thanks once again and please close the ticket!

bancika, 2022-11-06:
Cool!

closed 2022-11-06 by bancika