bancika/diy-layout-creator #740: Cannot edit component (electrolytic capacitor) on 4.32

alandmoore, 2022-11-13:
With 4.32 on Arch Linux using openjdk 11.0.17, I encountered a bug where I could not edit or delete an electrolytic capacitor.  Any time I tried to edit it, I got an error dialog stating that an error had happened and to check the log.  Trying to delete the component had no effect.

I found the following errors on the console each time I tried to do this:

```
15:03:16,991 DEBUG [AWT-EventQueue-0] Presenter - dragStarted(java.awt.Point[x=1640,y=1641], 2)
Exception in thread "AWT-EventQueue-0" java.lang.RuntimeException: java.lang.CloneNotSupportedException: Could not clone the component. Reason: null
        at org.diylc.core.Project.clone(Unknown Source)
        at org.diylc.presenter.Presenter.dragStarted(Unknown Source)
        at org.diylc.swing.plugins.canvas.CanvasDragGestureListener.dragGestureRecognized(Unknown Source)
        at java.desktop/java.awt.dnd.DragGestureRecognizer.fireDragGestureRecognized(DragGestureRecognizer.java:361)
        at java.desktop/sun.awt.X11.XMouseDragGestureRecognizer.mouseDragged(XMouseDragGestureRecognizer.java:215)
        at java.desktop/java.awt.AWTEventMulticaster.mouseDragged(AWTEventMulticaster.java:327)
        at java.desktop/java.awt.AWTEventMulticaster.mouseDragged(AWTEventMulticaster.java:327)
        at java.desktop/java.awt.AWTEventMulticaster.mouseDragged(AWTEventMulticaster.java:327)
        at java.desktop/java.awt.Component.processMouseMotionEvent(Component.java:6683)
        at java.desktop/javax.swing.JComponent.processMouseMotionEvent(JComponent.java:3360)
        at java.desktop/java.awt.Component.processEvent(Component.java:6404)
        at java.desktop/java.awt.Container.processEvent(Container.java:2263)
        at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5011)
        at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
        at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843)
        at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918)
        at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4564)
        at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488)
        at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307)
        at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772)
        at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843)
        at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
        at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
        at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
        at java.base/java.security.AccessController.doPrivileged(Native Method)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
        at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
        at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
        at java.base/java.security.AccessController.doPrivileged(Native Method)
        at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
        at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
        at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
Caused by: java.lang.CloneNotSupportedException: Could not clone the component. Reason: null
        at org.diylc.components.AbstractComponent.clone(Unknown Source)
```
(Could not actually locate the diylc log file on linux, but I can get a more complete console output if required).

Downgrading DIYLC to 4.31 resolved the issue for now.

bancika, 2022-11-13:
can you please upload a diy file that stopped working? I tried making one
from scratch and it worked fine.

alandmoore, 2022-11-13:
[problem-file.zip](https://github.com/bancika/diy-layout-creator/files/9998488/problem-file.zip)

Here's the problem file.  If I open it, drag out an electrolytic cap, and try to edit it, I get the error.  I do have a custom config, which I've also attached.  Not sure if that is part of the issue.

[my-diylc-config.zip](https://github.com/bancika/diy-layout-creator/files/9998503/my-diylc-config.zip)

Beamer4U, 2022-11-15:
I'm operating Linux Mint 19, please see the image for details:
[image]
With DIY Layout Creator v4.32.0 I'm now unable to edit. I can select a component as: 
[image]
but that is it - absolutely no more editing! I was able to build this layout over a number of occasions, then no more editing! I now can't even edit those earlier layout occasions.

bancika, 2022-11-15:
> [problem-file.zip](https://github.com/bancika/diy-layout-creator/files/9998488/problem-file.zip)
>
> Here's the problem file. If I open it, drag out an electrolytic cap, and try to edit it, I get the [...]

Thank you for the information. I tried that file with your config and worked fine on my machine. Maybe it has to do with Linux, would need to try on some machine. Will respond when I find something.

@Beamer4U I understand that you are frustrated but you can always go back to the previous version. DIYLC comes with no warranty. It's just me doing all the work, it's impossible to try all the features on all possible OS's....so just chill down :)

bancika, 2022-11-15:
thank you for the follow-up and the clarification. So far I received few complaints and in all cases version 4.31 worked where 4.32 didn't. I tried @alandmoore files on windows, mac, mint 20 and ubuntu 19 and they worked fine...there's something I'm missing to catch the issue.

What would be of tremendous help is a step by step guide to make it fail, from the moment you open the app. That way I can be sure to replicate those steps

alandmoore, 2022-11-15:
@bancika For me it's as simple as this:

- Open DIYLC 4.32
- Drag an electrolytic cap onto the blank canvas
- Edit the value
- Click OK.  Error message pops up.

I tried this with a default config it does the same.  Perhaps it's java version related?  Here's the output of `java --version`:

```
Picked up _JAVA_OPTIONS: -Dawt.useSystemAAFontSettings=gasp
openjdk 11.0.17 2022-10-18
OpenJDK Runtime Environment (build 11.0.17+1)
OpenJDK 64-Bit Server VM (build 11.0.17+1, mixed mode)
```

alandmoore, 2022-11-15:
Ok, no idea why I was using such an old version of Java, but I've switched openjdk 19 and it's the same story.

Here's the version info for that environment:

```
Picked up _JAVA_OPTIONS: -Dawt.useSystemAAFontSettings=gasp
openjdk 19.0.1 2022-10-18
OpenJDK Runtime Environment (build 19.0.1+10)
OpenJDK 64-Bit Server VM (build 19.0.1+10, mixed mode)
```

bancika, 2022-11-15:
Thanks for the info. I ruled out java version because I have one report that uses java 8, which is the minimal required version.

Beamer4U, 2022-11-15:
I hope this is clear and helps: RIAA-23NewDes2 was largely developed in a prior version (I think v4.10?), then was modified in v4.32 - no problems. Components were then stripped off, and a mirror image was done to achieve a cut pattern board (RIAA-23NewDes2a) - no problems. Modifications were then made to RIAA-23NewDes2 (components in green) - a few values, and part number changes. That's when the trouble started - no more editing of RIAA-23NewDes2  available on v4.32.

Files:
DIYLC file in TXT format and a photo of the same file.

RIAA-23NewDes2.txt failed file:
[RIAA-23NewDes2.txt](https://github.com/bancika/diy-layout-creator/files/10016811/RIAA-23NewDes2.txt)
RIAA-23NewDes2 photo:
[image]

RIAA-23NewDes2a.txt working file:
[RIAA-23NewDes2a.txt](https://github.com/bancika/diy-layout-creator/files/10016845/RIAA-23NewDes2a.txt)
RIAA-23NewDes2a photo:
[image]

Thank you for your thoughts.

alandmoore, 2022-11-16:
Dumb question, but how do I compile diylc from source?  Sorry, my java skills are minimal and 15 years out of date.  The try/except blocks in AbstractComponent.clone are masking which line is actually throwing the error, so I wanted to try running it without them to get some clarity.

alandmoore, 2022-11-16:
Still working on being able to compile this, found some more interesting things:

- It only affects the Electrolytic caps as far as I've found
- It happens on my Raspberry Pi running raspian.  Used the downloaded zip file, just like I do on arch.
- The exception thrown was different sometimes, showing the following: 
   ```
Caused by: java.lang.CloneNotSupportedException: Could not clone the component. Reason: wrong number of arguments
        at org.diylc.components.AbstractComponent.clone(Unknown Source)
```

bancika, 2022-11-16:
@alandmoore don't bother trying to build it...this is actually helpful, I haven't seen this error log before in any of the bug reports.
I have a clue, will build a test version today for you to try.

bancika, 2022-11-16:
okay, please try this [https://wetransfer.com/downloads/71bd4f8e6231d79c55b0d363df3c3df720221116071957/87833768640fdc4dff16649b28c003f720221116072013/a3c69f](https://wetransfer.com/downloads/71bd4f8e6231d79c55b0d363df3c3df720221116071957/87833768640fdc4dff16649b28c003f720221116072013/a3c69f?trk=TRN_TDL_01&utm_campaign=TRN_TDL_01&utm_medium=email&utm_source=sendgrid)

junagu, 2022-11-16:
I had the same issue, installed on MX-Linux. 
The download in the above post solved the issue.

bancika, 2022-11-16:
please try version 4.33

alandmoore, 2022-11-18:
Hi, sorry I hadn't responded yesterday.  4.33 fixes the issue for me.  Thanks!

closed 2022-11-18 by alandmoore