bancika/diy-layout-creator #741: Switch not working

mrkeperkins, 2022-11-15:
I have tried using switches, but they do not seem to work. I'm not sure what issue is. Log doesn't show anything. Can labels be added to switch? I wired them with the C in the middle like most DPDT and it didn't work. Not sure where issue is.

bancika, 2022-11-15:
You have to be more specific. "Not working" can mean anything. Please
provide an example, diy file or a screenshot if the wiring is simple. What
are you trying to do? What doesn't work for you?

mrkeperkins, 2022-11-15:
The switching didn't work. It acted like an open circuit. I found by dumb
luck if you move the switch away from potentiometer, it works. I guess if
you put the switch on the pot to make it appear like a push/pull pot it
connects lugs of switch to pot - which I had grounded.

Hope this helps

bancika, 2022-11-15:
Okay, I got it but only because you said that it acts as an open circuit :)
Yes, the back of the potentiometer is considered as a conductive surface so
it causes all terminals of the switch to short. Will try to fix it soon.

Going forward please be more specific. "Not working" can apply to rendering
on the screen or editing a component, maybe exporting to a file...I can't
possibly know what you are looking at on your screen and trying to
accomplish.

bancika, 2022-11-26:
Fixed for 3.34

closed 2022-11-26 by bancika