bancika/diy-layout-creator #742: 4.32.0 - OSX can't move components on existing layout

chipshorter, 2022-11-15:
I upgraded from 4.2x.0 (sorry I'm not sure which, but was at least 6 months old) to 4.32.0 and found that although I could open old layouts, and could select components and edit properties, I could not move them, stretch them etc. This applied to all components I tried including boards.

I was able to create a new layout and add and move components fine.

I decided to install 4.31.0 to see if that was the same, and that appears to work OK, so it looks like 4.32.0 broke something on OSX

bancika, 2022-11-15:
Hi,
can you please upload few of the logs from around the time you had issues? You can find them in ~/diylc/logs and each is timestamped.
Thanks

bancika, 2022-11-16:
I have prepared a build with something that might help with your issues, can you please try it?
[diylc-4.32.0.zip](https://we.tl/t-56jo0jbmQX?utm_campaign=TRN_TDL_05&utm_source=sendgrid&utm_medium=email&trk=TRN_TDL_05)

chipshorter, 2022-11-16:
This is the log from when I was playing with 4.32.

I'm going to try your special build shortly and will report back
[log-2022-11-15 01.44.15.txt](https://github.com/bancika/diy-layout-creator/files/10025298/log-2022-11-15.01.44.15.txt)

chipshorter, 2022-11-16:
Hmm, maybe I did something wrong I'm not sure, but it errored out on loading (see attached screenshot)
[image]

bancika, 2022-11-16:
please try version 4.33

chipshorter, 2022-11-17:
As far as I can tell (tried at least 5 old layouts) 4.33 solves the issue I had. Thanks for your effort in resolving it.

I did get a message on one layout "File was opened, but there were some issues with it" "The project references unknown component properties..." however, it appeared to work OK, so I'm not sure what caused that. If I bump into more of those I'll open another issue.

closed 2022-11-17 by chipshorter

bancika, 2022-11-17:
Can you please upload that project so I can examine if the change was
intentional?

chipshorter, 2022-11-18:
[si_fuzz_rite.diy.zip](https://github.com/bancika/diy-layout-creator/files/10041831/si_fuzz_rite.diy.zip)
There you go

bancika, 2022-11-18:
Thanks, I saw the issue...nothing to worry about.