bancika/diy-layout-creator #743: Some variants are not drawn/placed (but selectable) in 4.33 on Win10 Java 8 (1.8.0_281-b09)

pixelwaster, 2022-11-18:
I am trying to create a Manhattan style layout and have run into a problem with saved variants. 
The specific ones are the square and round pads (.23" / ~6.4mm). The parts are made from blank board with a copper layer. Base piece is also a bare board with a lighter copper layer.indicated by the red text on the status bar at the bottom

I can right click Bare Board and select any of the variants. Only the pads aren't drawn when dragging and can't be placed indicated by the red text on the status bar at the bottom.

Files:
[manhattan by michael.xml.txt](https://github.com/bancika/diy-layout-creator/files/10044118/manhattan.by.michael.xml.txt)
[log-2022-11-18 12.18.05.txt](https://github.com/bancika/diy-layout-creator/files/10044119/log-2022-11-18.12.18.05.txt)
[test-manhattan.diy.txt](https://github.com/bancika/diy-layout-creator/files/10044121/test-manhattan.diy.txt)

and a sample project, Just created the pads in situ and copy-n-paste as needed.
[DIZ-amp.diy.txt](https://github.com/bancika/diy-layout-creator/files/10044130/DIZ-amp.diy.txt)
eded.

Great program, been using since you left Google.

bancika, 2022-11-18:
Can you please upload your config file and tell me which variants are
failing

bancika, 2022-11-18:
Also,  did this work prior to this version?

pixelwaster, 2022-11-19:
This is the first time I tried to make the Manhattan variants. I have other variants that I've made and downloaded, they all work in previous versions and 4.33

The non-working ones are Manhattan Square Pad and Manhattan Round Pad in BlankBoard

[config.xml.txt](https://github.com/bancika/diy-layout-creator/files/10046376/config.xml.txt)

bancika, 2022-11-20:
Thanks, was able to find the issue and fix it for the upcoming version.

Btw, what is Manhattan style layout?

closed 2022-11-20 by bancika

pixelwaster, 2022-11-21:
It's a technique of building circuits that lie between point-to-poimt/dead
bug/ugly and an etched PCB, There is a style called MUPPET promoted by
Chuck Adams that combines elements of all the different styles. There are
some Manhattan builders that make some absolutely beautiful projects,

Quoting k8iqy:
Manhattan-style construction is based on using small pads as connecting
points for circuitry that is built over a conductive substrate.  In most
cases, the conductive substrate is a piece of printed circuit board
material.  However, it could also be a piece of flashing copper, or
cardboard with an adhesive backed conductive and solderable material
affixed.

Pads are attached to the substrate using any of the commercially available
"super glues", which sets in seconds and will afford a very durable bond if
the substrate surface is clean and free of grease or oil. This method is
similar to "ugly construction" where the parts are soldered together and to
the substrate, but more organized and easier to modify if the design needs
changing, or to trouble-shoot if problems arise.

bancika, 2022-11-21:
The issue you reported is fixed but now I see other issues with this approach and the way things currently work in DIYLC.
If you place a small copper board on top of a large copper board, both surfaces will be considered as connected because DIYLC does not consider the layer of insulator between the two copper layers.

pixelwaster, 2022-11-22:
Took some time to understand what you meant. Then I remembered the other
options under File-> Analyze. The Manhattan pads are BareBoards with the
color changed, there is no copper clad board option. The pads have no
connectivity. For now, I'm happy the issue is fixed. Maybe I should learn
Java??

As an experiment, I did make a Solder Pad 6mm square with a zero sized
hole, The Expand Selection -> All Connected did highlight the pads and
resistors on a test layout, The copper colored BareBoard was "ignored" in
both types of netlists. There was a jack in the "circuit" but the analyze
guitar  could not find anything connected to a jack tip.

If the 6mm Solder Pads were used over the Ground Fill they were seen as a
single node in the netlist.

test file attached.

<?xml version="1.0" encoding="UTF-8" ?>
<project>
  <fileVersion>
    <major>4</major>
    <minor>33</minor>
    <build>0</build>
  </fileVersion>
  <title>New Project</title>
  <author>Michael</author>
  <width value="11.0" unit="in"/>
  <height value="8.5" unit="in"/>
  <gridSpacing value="0.1" unit="in"/>
  <dotSpacing>1</dotSpacing>
  <components>
    <diylc.connectivity.SolderPad>
      <name>Pad2</name>
      <size value="0.24" unit="in"/>
      <color hex="da8a67"/>
      <point x="8.3" y="6.5"/>
      <type>SQUARE</type>
      <holeSize value="0.0" unit="mm"/>
      <layer>_5</layer>
    </diylc.connectivity.SolderPad>
    <diylc.connectivity.GroundFill>
      <name>GF1</name>
      <controlPoints>
        <point x="7.7" y="6.3"/>
        <point x="7.7" y="7.5"/>
        <point x="9.195" y="7.5"/>
        <point x="9.195" y="6.3"/>
      </controlPoints>
      <color hex="000000"/>
      <pointCount>_4</pointCount>
      <layer>_5</layer>
    </diylc.connectivity.GroundFill>
    <diylc.boards.BlankBoard>
      <name>Board1</name>
      <alpha>127</alpha>
      <value></value>
      <controlPoints>
        <point x="5.8" y="5.8"/>
        <point x="7.4" y="6.0"/>
      </controlPoints>
      <firstPoint x="5.8" y="5.8"/>
      <secondPoint x="7.4" y="6.0"/>
      <boardColor hex="da8a67"/>
      <borderColor hex="000000"/>
      <coordinateColor hex="b6b6b6"/>
      <xType>Numbers</xType>
      <coordinateOrigin>Top_Left</coordinateOrigin>
      <coordinateDisplay>One_Side</coordinateDisplay>
      <yType>Letters</yType>
      <length value="40.0" unit="mm"/>
[... 180 lines cut ...]
      <name>J2</name>
      <alpha>127</alpha>
      <points>
        <point x="7.6" y="6.6"/>
        <point x="7.6" y="6.4"/>
        <point x="7.6" y="6.5"/>
      </points>
      <bodyColor hex="ffffff"/>
      <borderColor hex="000000"/>
      <labelColor hex="000000"/>
      <leadColor hex="0000ff"/>
      <display>NAME</display>
      <flipStanding>false</flipStanding>
      <labelOriantation>Directional</labelOriantation>
      <moveLabel>false</moveLabel>
      <hideShortLeads>false</hideShortLeads>
      <style>SOLID</style>
    </diylc.connectivity.Jumper>
  </components>
  <groups/>
  <lockedLayers/>
  <lockedComponents/>
  <hiddenLayers/>
  <font name="Arial" size="14" style="0"/>
</project>