bancika/diy-layout-creator #745: Custom Component using json config files

leonllrmc, 2022-11-22:
Hi, I am a student at the CFPT electronic school, here we use Diylc but sometime we need to use a component that isn't in the software so we have to create it with a mix of shapes and existing components, I heard my teacher dissusing about the fact that they would like to add the components they need but that it would be not easy because it need Java, so that would be a really cool idea to be able to create custom components from a json file

bancika, 2022-11-26:
Hey there!

Out of curiosity, where is your school?

There used to be similar functionality in DIYLC gen2 where all components were defined in XML. I ultimately abandoned that approach because it's hard to model all behavior of components without the code. Besides the look, components also have functional aspects - some act as switches, some as connectors, some as planes of continuity, etc. Also, they have user editable properties that affect how they look and work. All that is hard to accomplish without the code.

In theory, it would be possible to do something with JSON that would be only for visual representation of a component...but I estimate it's not a simple feat. I wonder how much use it would see..

leonllrmc, 2022-11-27:
Hey, my School is in Geneva, In Switzerland.
The custom components would be used only for graphical representation, they would not need to be as customisable and complex as standalone components that would use Java, but they would help to have concrete components, I can't find a good way to explain.
They would only need to have a texture, dimensions, and x and y (in inches or meters) for where the pins are.
I would have loved to show you an example of the workaround they use, but I forgot my USB stick, but for example, my teacher used a square and some dots to represent a button, or one time, we used a LED to represent a LDR.

bancika, 2022-11-27:
In that case, building block functionality inside diylc is the right
solution for you. Just create the complex shape, select all components and
save as a building block. You can then use it as a component from the left
side menu

leonllrmc, 2022-11-27:
Thanks a lot for your help, and bye.

closed 2022-11-27 by leonllrmc

bancika, 2022-11-27:
You're welcome!

You can also export building blocks to a file import them from a file, so
you can share them with your team mates...