bancika/diy-layout-creator #747: Tabbed Toolbar Obscured on Linux

Axldeziak, 2022-12-09:
Since version v4.31.0 the Tabbed Toolbar has been broken when running the program on Linux. This applies to the versions using run.sh to start, as well as the flatpack versions.
The issue is the toolbar does not open fully and the tabs are half obscured.  v4.31, 4.32, 4.33, and 4.34 all do this. I have been stuck using version v4.30.1 just to avoid it.

[image]

[image]
.

bancika, 2022-12-10:
In the meantime you can use component tree instead of the toolbar, it's more convenient anyways.

Axldeziak, 2022-12-11:
No, the component tree is not more convenient for me. I find tabbed to be much easier to use and less intrusive.
(It throws off off your train of thought when 20% of the screen is suddenly taken over and everything flies to the right while you root through the options like your digging through the junk drawer. And then everything flies back to the left when you close the drawer. It takes me completely out of what i'm doing and is just jarring.)

Regardless, thanks for taking the time to look into the issue. I do appreciate it.

Axldeziak, 2022-12-11:
The resolution on this laptop is 1366X768 (the largest it will go) on a 15 1/2" screen. It's my daily driver and works quite well so I have no need for a desktop with a jumbotron tv attached.
Please keep the tabbed toolbar, it does get used, and is quite useful.