bancika/diy-layout-creator #751: Wire curve is lost when applying variant to an existing wire

joeSeggiola, 2023-01-09:
I work with "Hookup Wires" a lot, and I noticed that if you apply an existing variant to an existing wire (for example, to change its color from a user-defined list) the curve given to the wire is lost, as if all the additional points are re-set in the middle of the two extremes. Also, possibly related, this happens also when saving a variant from an existing wire, so that it is impossible to save and retain the curve of a wire as a variant.

This made me think that it is strange that the default wire is created with a single click and a "default" length and curve. It would be probably better to create them with two click as "Copper Traces" and "Jumper", so that start and end points can be quickly entered. At the moment wire creation is a bit awkward, as you have to click the start point, the search for the end point (that can be quite far off when using high zoom levels), to click and drag it in place...

bancika, 2025-02-10:
The problem with wires is that they have 4 (or more) control points for the bezier curve control...so I cannot create them with two clicks like two-point components. Selecting the first point is ok, but then for the second click I would have to improvise the location of the middle two points while you move the 4th point around. It's not easy :/

bancika, 2025-03-28:
I found a bug and fixed it for 5.2.0

closed 2025-03-28 by bancika