bancika/diy-layout-creator #752: 4.35 really slow/laggy on Windows box

mhuss99, 2023-01-13:
This evening I updated to 4.35. When I run it, all editing is really laggy (seconds). The left hand menu and menu bar were blank until I ran my mouse cursor over them. Double-click on an item, wait, edit, save, wait... Drag and drop is a nightmare!

I went back to version 4.34, which is working fine. New video lib or something?

Win10 Pro x64, older 3.5GHz i7, 16GB, two 1920x1080 monitors, NVIDIA GeForce GT 710 (not a gamer, lol)

mhuss99, 2023-01-13:
I fired it up empty and noticed the menu/side bar issues, so I restarted it, and everything looked OK. Then when I loaded a project to work on I noticed the editing and drag/drop laggy behavior.

bancika, 2023-01-15:
There were no changes that could lead to that behavior, strange

pixelwaster, 2023-01-15:
I finally got chance to use the new version and can confirm it is slow. New project. Seems to be operating fine. Grab any random part or symbol and it gets slow, Sometimes is recognizes the ESC to stop continuous add.  By my third part place the screen went black (during a refresh?)

JohnEdF, 2023-01-29:
I've got the same problem as pixelwaster

bancika, 2023-01-29:
Does 4.34 work fine for you in comparison with 4.35?

JohnEdF, 2023-01-29:
I upgraded from 4.31 to 4.35 and had the problem. I've just upgaded to 4.34 (and it also updated Java) and it worked fine. Upgraded to 4.35 again and it now works fine - hopefully it was a Java problem

bancika, 2023-03-29:
I found few issues that could cause this. If you were using radial inductor or fuse symbol, they can cause performance degradation.

I fixed for 4.36

closed 2023-03-29 by bancika