bancika/diy-layout-creator #753: OS x issue - main project pane just... not working?

thatdavejordan, 2023-01-18:
Hey guys, all of a sudden this morning the main window in DIYLC just kind of... stopped registering mouse clicks?  i tried restarting my machine, tried rolling back a couple versions, tried my bluetooth mouse, tried the trackpad on my laptop, nothing with real results.  it will sometimes place items but then won't let me select them or move them or edit them, and every so often if i click like 30 times it'll register one click, but something is definitely up.  any ideas?

MarsManiac, 2023-02-23:
Now happening to me on Mac with Ventura 13.2.  DIYLC main window is running glacially slow in adding/deleting objects, registering changes onscreen.  Select an object and wait 15 seconds before it turns red.  Delete it and wait another 15 seconds for the screen to register the change.  Activity monitor shows nothing eating up CPU cycles.  DIYLC is...just...slow.

bancika, 2023-02-23:
Does the previous version work the same?

MarsManiac, 2023-02-23:
Sorry, false alarm.  I downloaded a .diy file from an online pal.  That file updated very slowly.  After a few hours, I discovered that all my own DIYLC files worked fine, the app was as quick as ever.  Something about his file throttled DIYLC down really badly.  But all my own stuff works fine, so there must be something in that download file.  Please ignore my comments.  4.35 is working for me just fine.  Except I can’t find that Title Block. ;-(

Thanks,

MarsManiac

bancika, 2023-03-29:
Title block fixed for 4.36

closed 2023-03-29 by bancika