bancika/diy-layout-creator #756: New version problems. 4.35.0

MarsManiac, 2023-02-11:
Downloaded v4.35.0 recently, today first time using it.  Some new issues appeared:
1.  Favorites tab is empty now.  I had *many* favorites set up. 
2. Recents tab clears when app closes (maybe by new design but I liked it the old way)
3. Many building blocks I had created are now *gone*. 
4. Maybe related?  The new building block: Title Box  doesn't appear anywhere in the elements that I can find.

Is is by design that my self-created elements are gone?  Is there any way I can get them back?  I did nothing with DIYLC at all since before Christmas, didn't even start it up, until updating to 4.35.0 on January 23, 2023.

Thanks!

bancika, 2023-02-11:
That typically happens if something in your config file is not compatible
with the latest version. That's not intentional but happens. Your old
config file should be backed up in ~/diylc folder, you can rename it back
to config.xml and run older versions to verify. Once you locate your good
config, please send it to me to find what went wrong so I can fix it.

MarsManiac, 2023-02-11:
Excellent!  I will do that and report back.  Thanks!

MarsManiac, 2023-02-14:
> Excuse me if I’m doing this wrong or looking at it wrongly.   I believe I’m doing as you asked, but so far not getting a helpful result.
>
> After an hour’s work, I cannot find any combination of app files / old config.xml files that will show my Favorites and my additions to Building Blocks.
>
> What I’ve been doing is hunting for a prior version of DIYLC that will work with my *bad* config.xml file?  Is that correct?
>
> If it would help, I can send you the ‘bad’ config.xml file, the one that failed to work with the v4.35.0 upgrade.

I forgot to mention one important point in this last email.  The prior version of DIYLC that I used was 4.32.0.  It did not work with my config.xml~ that I had copied back in.  It did not throw any error, but my personal objects were still missing from the app when I ran it.  Here it is, in case it helps.

￼

MarsManiac, 2023-02-15:
> What I meant was - try to copy the latest (and probably the largest) backup file back to config.xml and run the previous version that runs fine. Alternatively, you can open the config file and search for some of your data...it's an XML and fairly straight forward to read. Just to verify that it's the one. Once you get it, please send it over. Thanks

I think I’ve followed your instructions but still don’t have a conclusive answer for you.  The largest config file I have anywhere (175K) is attached below.  This version doesn’t seem to have my objects in it either and I’m comfortable reading XML.

The last version I ran before 4.35.0 was 4.32.0.   4.32.0 and this config file did not run; the app popped the error message and replaced this file with the default version.

At this point, feel free to mark it closed.  If there is no resolution to this, I’m fine with that.  I haven’t lost anything irreplaceable and I’m sure you have better things to do. I’ll just watch my update installs more closely and make sure I have better data for you if the problem recurs.

Just an idea: Could you write a timestamp to the xml file at update time? 

Thanks for you help and for making a great app!

￼

MarsManiac, 2023-02-23:
Hi,

I’m adding to the old ticket because this seems related.

I finished up a project earlier and just downloaded vers. 4.35 again.  This time it preserved my Recently Used so that’s good.  I can find most of the other added objects in 4.35, but not the new Title Block, not on Building Blocks or elsewhere.  Can you suggest why?  

Thanks

Richard White

PenBach, 2023-03-13:
I just upgraded to v4.35 from v4.29.0 and wish I hadn't.
My "Recent Files" were lost.
The config was replaced with a default.
When I opened the last layout that I worked on, the Lever Switch - Super Switch was not as it should be (was before). Not sure if there was anything else out of place; that's the one that lept out as being wrong.

bancika, 2023-03-13:
Can you try the same as I asked the other reporter? There should be a backup of your config file in ~/diylc directory. The most recent backup or the largest size is probably your old config. I'd like to see it so we can figure our why the latest version fails to load it. If you find it you can rename it back to config.xml and use the old version of diylc as you did before

MarsManiac, 2023-03-13:
I did send it once before on 2/23.  Here’s the text of that email again in case you didn’t get it.

I think I’ve followed your instructions but still don’t have a conclusive answer for you.  The largest config file I have anywhere (175K) is attached below.  This version doesn’t seem to have my objects in it either and I’m comfortable reading XML.

The last version I ran before 4.35.0 was 4.32.0.   4.32.0 and this config file did not run; the app popped the error message and replaced this file with the default version.

At this point, feel free to mark it closed.  If there is no resolution to this, I’m fine with that.  I haven’t lost anything irreplaceable and I’m sure you have better things to do. I’ll just watch my update installs more closely and make sure I have better data for you if the problem recurs.

Just an idea: Could you write a timestamp to the xml file at update time? 

Thanks for you help and for making a great app!

￼

bancika, 2023-03-13:
I was referring to @penBach, he complained earlier today

MarsManiac, 2023-03-13:
My mistake.  Anyway, just sent you another follow-up.  That config file example I sent you seems to be badly terminated.  Look at the last line.  NRN

bancika, 2023-03-13:
Oh, could be that the app quit unexpectedly while writing to the file.

MarsManiac, 2023-03-13:
Quite possibly, although to me the upgrade of 4.35 seemed to go well.  Hope you can fix this but I’ll be backing up my config file from now on, just in case.

Thanks!

PenBach, 2023-03-14:
>There should be a backup of your config file in ~/diylc directory.

Could you tell me where I can find the backup config?
"~/diylc" does not seem like a directory to me.

I can see the config.xml file in directory "C:\Program Files\DIYLC", but there's no backup files there.
I'm on WinXP BTW.

bancika, 2023-03-14:
~ designates user directory. It should be in the directory C:\Documents and
Settings

PenBach, 2023-03-14:
Here's my backup config.
I had to change the extension in order to upload here. I'm sure you know what to do.
If it doesn't come out right, is there a better way to send it to you?
[config.xml.bak.txt](https://github.com/bancika/diy-layout-creator/files/10969104/config.xml.bak.txt)

bancika, 2023-03-29:
@PenBach the config you uploaded loads up fine, but it only has few of your settings and recent files. I'm guessing that's a backup from after the crash.

I found an issue that can happen in rare cases that makes the config file corrupt by trimming it short. It should be fixed in 4.36

Also fixed performance issues caused by some components.

closed 2023-03-29 by bancika