bancika/diy-layout-creator #757: Nothing move anymore

BenWaSoul, 2023-02-15:
Hi. Recently installed on Win10 64bits, The load is longer than before and items don't move. I can draw wires but can not move their point ???? nothing can be moved.

bancika, 2023-02-15:
I need some more information. Does it happens with a blank project as well?
Do you have something in the Configuration menu? Highlighted connected
areas and maybe other options lock the drawing. Also there are options
under Layers menu for explicit locking of a layer.

BenWaSoul, 2023-02-15:
Blank project as well yes. All Layers are Visible not locked. I don't understand the change. Surly my config but I don't know what... Tried to reinstall. The same. 

Does it need Framework.net FX 3.5 SP1 ? I have recently install it for an old software. No other change.

bancika, 2023-02-15:
Can you try to rename config.xml file temporarily, it's in ~/diylc folder.
That will reset all the config to default. You can switch the files back
after.

.net doesn't affect diylc.

BenWaSoul, 2023-02-15:
mmm I don't have config.xml anymore ???

BenWaSoul, 2023-02-15:
I have no sliding selection. The icon change to finger on wires point but clic does nothing.

bancika, 2023-03-30:
Does it still happen with 4.36? If it does, please send over the log from the time it happens. It's in c:/users/yourName/diylc/logs

BenWaSoul, 2023-04-27:
Hi Bancika. How are you? 

I encounter the same trouble with 4.36 and the latest 4.37 version as well on Win10. I can move items with the keyboard but not with the mouse, so I am not able to wire anything... I would notice I use DIYLC on Win7-64 on another old PC and everything is fine, nothing to complain on this plateform.

I send you the log file :

[log-2023-04-27 07.30.29.txt](https://github.com/bancika/diy-layout-creator/files/11346451/log-2023-04-27.07.30.29.txt)

bancika, 2025-02-20:
@BenWaSoul is this still an issue? If yes, can you please make a copy of your config.xml file, rename it to config.txt and upload it here? You can find it in c:/Users/<yourname>/diylc folder

BenWaSoul, 2025-02-21:
Hey @bancika ! Thank you for your interest. 

Still the same issue even with the 4.39 on Win10_64, all the options disable or enable. 

I noticed that the "highlight connected areas" option enabled change the cursor to a cross with what it's impossible to select anything...

[config.txt](https://github.com/user-attachments/files/18914706/config.txt)

bancika, 2025-03-03:
Hi. I tried to load up your config and my 4.39 still works without issues. I wonder what could be the issue. Can I ask you for one last favor? If you could record a quick video - starting from a blank project, how you get it to the state where nothing moves? Maybe it would help me understand.

bancika, 2025-03-03:
> Hi Bancika, TY.
>
> It is very simple : I open DIYLC from the shortcut. The animated splash
> screen loads but takes 15~20s to open the window. I take an item from the
> list, slide it to the sheet. It is displayed normally, but at this step, I
> can't move it with the mouse.
> I can only select/deselect it by clicking, and move it with the keyboard.
> So, unable to catch wire-dots to morph it.
>
> I'm stuck.

When you move the mouse over a component, what is the cursor showing?
Arrow, hand or a crosshairs?

Thanks

BenWaSoul, 2025-03-03:
Normal arrow over the sheet, then hand over items. I am not able to make a video but I will try later.

Is it possible that a service of Windows or a security suit (Norton) can cause this issue ?

bancika, 2025-03-26:
shouldn't be security issue

BenWaSoul, 2025-07-06:
Hello Bancika. I would like to mention that the  diylc-5.5.0-win64.exe  is perfectly working, all is functional, and I can move everything now. 

TYSM for the good job. This thread can be closed.

closed 2025-07-07 by bancika