bancika/diy-layout-creator #760: 4.35 *vewry* slow on MacBook Pro Ventura 13.2

MarsManiac, 2023-02-23:
DIYLC main window is running glacially slow in adding/deleting objects, and generally registering changes onscreen.  Select an object and wait 15 seconds before it turns red.  Delete it and wait another 15 seconds for the screen to register the change.  Activity monitor shows nothing eating up CPU cycles.  DIYLC is...just...slow.

MarsManiac, 2023-02-23:
It turned only to be happening on one file, downloaded from a forum.  Not sure why it's bad, but all my own files still work fine and DIYLC is as quick as ever.  Closing this issue.

closed 2023-02-23 by MarsManiac

bancika, 2023-02-24:
can you please upload the file that causes you issues (you might need to
rename it to .txt instead of .diy). I need to investigate the source of
slowness.

MarsManiac, 2023-02-24:
Here you go.  Good hunting!

￼

-- 
Branislav Stojkovic
www.diy-fever.com
—

<?xml version="1.0" encoding="UTF-8" ?>
<project>
  <fileVersion>
    <major>4</major>
    <minor>27</minor>
    <build>0</build>
  </fileVersion>
  <title>New Project</title>
  <author>Shane</author>
  <width value="11.0" unit="in"/>
  <height value="6.0" unit="in"/>
  <gridSpacing value="0.1" unit="in"/>
  <dotSpacing>1</dotSpacing>
  <components>
    <diylc.electromechanical.PhonoJackSymbol>
      <name>J1</name>
      <value></value>
      <controlPoints>
        <point x="0.9" y="1.3"/>
        <point x="0.9" y="1.5"/>
        <point x="0.9" y="1.4"/>
        <point x="0.5" y="1.4"/>
      </controlPoints>
      <color hex="000000"/>
      <flip>NONE</flip>
      <display>NAME</display>
      <orientation>DEFAULT</orientation>
      <moveLabel>false</moveLabel>
      <type>MONO</type>
    </diylc.electromechanical.PhonoJackSymbol>
    <diylc.electromechanical.PhonoJackSymbol>
      <name>J2</name>
      <value></value>
      <controlPoints>
        <point x="0.9" y="1.9"/>
        <point x="0.9" y="2.1"/>
        <point x="0.9" y="2.0"/>
        <point x="0.5" y="2.0"/>
      </controlPoints>
      <color hex="000000"/>
      <flip>NONE</flip>
      <display>NAME</display>
      <orientation>DEFAULT</orientation>
      <moveLabel>false</moveLabel>
      <type>MONO</type>
    </diylc.electromechanical.PhonoJackSymbol>
    <diylc.misc.GroundSymbol>
      <name>GND1</name>
      <point x="0.9" y="2.2"/>
      <color hex="000000"/>
      <size value="0.15" unit="in"/>
      <type>DEFAULT</type>
      <orientation>DEFAULT</orientation>
    </diylc.misc.GroundSymbol>
    <diylc.misc.GroundSymbol>
      <name>GND2</name>
      <point x="0.9" y="1.6"/>
      <color hex="000000"/>
      <size value="0.15" unit="in"/>
      <type>DEFAULT</type>
      <orientation>DEFAULT</orientation>
[... 1881 lines cut ...]
      <name>LN3</name>
      <alpha>127</alpha>
      <points>
        <point x="8.4" y="4.8999999999999995"/>
        <point x="9.5" y="4.8999999999999995"/>
        <point x="0.0" y="0.0"/>
      </points>
      <bodyColor hex="ffffff"/>
      <borderColor hex="000000"/>
      <labelColor hex="000000"/>
      <leadColor hex="cccccc"/>
      <display>NAME</display>
      <flipStanding>false</flipStanding>
      <labelOriantation>Directional</labelOriantation>
      <moveLabel>false</moveLabel>
      <hideShortLeads>false</hideShortLeads>
      <color hex="000000"/>
      <style>SOLID</style>
      <thickness value="1.0" unit="px"/>
      <arrowSize value="5.0" unit="px"/>
      <arrowTx serialization="custom">
        <java.awt.geom.AffineTransform>
          <default>
            <m00>1.0</m00>
            <m01>0.0</m01>