bancika/diy-layout-creator #761: is a Debian (.deb) install available anywhere?!

robert-rozee, 2023-03-05:
i'm running Linux Mint, and the latest 'normal' install available through the Mint Software Manager is 4.8. for the latest version (4.35) the only option available is a "FlatPak" which is not a realistic path for me - it will consume most of the free flash space available on the netbook i use.

is there either a .deb package for 4.35 available anywhere, or a simple set of instructions on how to install using the sources? as far as i can tell, said instructions should be relatively easy, but i'm far from a Linux expert and loathe to 'go it alone' for fear of breaking something. version 4.8 actually does most of what i want, but i'd rather like to get the 'a new version is available' popup to go away!

cheers,
rob   :-)

M0JXD, 2023-03-05:
There isn't a .deb package but there is a universal build for Unix systems.

Follow the details under the "Universal" section here: https://github.com/bancika/diy-layout-creator/blob/wiki/Manual.md#installing-the-app

Basically, download the .zip and run the the "run.sh" bash script inside. You'll have to use the chmod command to get Linux to run it.

M0JXD, 2023-03-05:
Forgot to mention, you will need to have installed a Java Runtime Environment too if you haven't already.

Here are a few links that might help:

https://www.fosslinux.com/49964/install-java-on-linux-mint.htm
https://linuxhint.com/install_java_linux_mint/
https://www.java.com/en/download/help/linux_x64_install.html#install

robert-rozee, 2023-03-05:
excellent - it works! from here i can now see how to slot it in where the old 4.8 was installed.

cheers,
rob   :-)

closed 2023-03-26 by bancika