bancika/diy-layout-creator #764: MAC Book Pro M2 (64bits)

strateur, 2023-03-15:
Thank you for reporting issues and contributing ideas and features to DIYLC repository!

Before you proceed, please read the following:

* One request or bug report per issue, please. No bundling.
* If you are reporting a bug, please attach the latest application log file together with description of steps that lead to the issue. They can be found in the user directory under `.diylc/logs` directory. On Windows 7 and later, it's located in `c:/Users/<your username>/.diylc/logs`. On Linux and MacOS, it's typically in `/Users/<your username>/.diylc/logs`.
* If you are having issues starting the application under MacOS, make sure that 3rd party apps are allowed on your Mac. This document covers how to do it - https://support.apple.com/en-us/HT202491.
* Try to look for similar unresolved issues that are already reported to avoid duplicates.
* Make sure that you are running the latest version of DIYLC.

strateur, 2023-03-15:
Is the last review 4.35 is compatible with new Mac Book Pro M2? T be sure before I start the software on it.

bancika, 2023-03-15:
No reason for it to not work, if the app bundle doesn't work you can always use the universal build that runs from the terminal (provided that you have java JRE)

closed 2023-03-26 by bancika