bancika/diy-layout-creator #765: V 4.35.0 really slow

MarkR1951, 2023-03-26:
Hi,

I had used several versions of DIYLC with great success. However, the latest version runs really slow. I have uninstalled the previous version, rebooted, and installed the latest. Still, very slow to use. What happened?

bancika, 2023-03-27:
Do you load any particular project or use any particular components? I need
something to narrow down the issue.

MarkR1951, 2023-03-27:
thanks for responding quickly. The application slows down dramatically after about 12 components (any type) are placed into the project.  At first I thought it was because I had brought a file from a previous version into the new one, but I just tried to start a new project in the new application and it happened again. My experience with the previous version was quite different, I could place over 50 components into a project and edit them with no lag at all.

My computer isn't especially powerful, but it ran the previous release with no difficulties at all.  The new one loads up find, and lets me start placing components, but quickly bogs down.

I really like this program, and would like to use some of the new features. Any idea where I can start troubleshooting?

Thanks!

closed 2023-03-27 by MarkR1951

bancika, 2023-03-27:
Can you please send over a file that gives you troubles? bancika at
gmail.com

MarkR1951, 2023-03-27:
Hi,

Thanks again for your help.

I'll attach two files: diylc_test is one I just created to verify the
problem this morning.

1W ps controls ts is the file I was working on yesterday when I was having
difficulties.  Same today.

Love the program, hope to get it working fast again!

bancika, 2023-03-27:
It doesn't work if you send them as email attachments to github. Either
need to send directly to me or upload to github page as zip

MarkR1951, 2023-03-27:
Apologies, didn't know the protocol. Files are attached here
[DIYLC slow files.zip.zip](https://github.com/bancika/diy-layout-creator/files/11081899/DIYLC.slow.files.zip.zip)

bancika, 2023-03-27:
Thanks! I'll give it a try

bancika, 2023-03-28:
Thanks again for the info. It seems that the Inductor component is to
blame. It works normally without it. Can you please try without it and
confirm?

I'll look for the root cause and fix it.

MarkR1951, 2023-03-28:
I removed the inductor from the file, but that didn't seem to clear the issue.  As a check, I installed the latest version of the app on a different computer. Same story, opens fine, but modifying or creating the layouts rapidly bogs down. I deleted the new app, rebooted, and installed the previous app version I had used (4.32) and everything ran quickly and smoothly.

bancika, 2023-03-29:
Found another culprit - fuse symbol also causes issues. With the fixes for
fuse symbol and inductor both of your files work fine on my machine

MarkR1951, 2023-03-29:
Success! Thanks a lot! With no fuse schematic symbol the file runs smoothly as V 4.35 on both machines! Contributing via PayPal now!

bancika, 2023-03-29:
Glad to hear that! And thank you for the donation. I will release the nee
version soon with the fixes.

cniesen, 2024-01-13:
@MarkR1951 and @bancika It would be nice to keep this issue open until it's fixed and close it with a link to the MR.  That's the usual practice.

In any case the issue where the "Toroidal Inductor" component slows down the application still exists in v4.37.0.  Thanks for posting the issue so I can avoid the component.