bancika/diy-layout-creator #770: not a bug, just a suggestion -- showing a board's dimensions while resizing

peterjosvai, 2023-04-20:
thank you, Bancika, for developing this amazing piece of software!
everyone's favorite tool on their PC :)

a suggestion:

when resizing a shape, 
its size could be displayed in the middle... while resizing....

foe example: you're setting up a PCB board...
you know its desired dimensions...
but you have to  make it by trial and error :) 

or by using the measurement tool to align to...

anyhow...
for shapes and boards it would be great to have this feature...

if we already have it and it's just me who didn't realize it, then pardon me :)

thank you VERY MUCH for developing this treasure!!

Peter

bancika, 2023-04-22:
as a workaround you can select sizing mode of a board to "explicit size"
and just type in the numbers in the properties dialog.

peterjosvai, 2023-04-22:
I didn't know about this possibility! thanks!!

bancika, 2025-12-31:
This is added, showing both dimensions and length where available (wires, leaded components, etc). Will be out in 5.9.0

[image]