bancika/diy-layout-creator #774: Aggregate move sometimes fails on 4.37 - Poss. related to #757?

MarsManiac, 2023-05-06:
After upgrading to 4.37, (on MacOS Ventura 13.3,) there seems to be some new limits to moving a group of selected objects.  In earlier versions I could select an entire drawing and move every object at one time, either with touchpad or arrow keys.  Now certain groups of objects won't move together.  For instance, I have a Fender LTP phase inverter stage that has to be moved in sections b/c it won't move as a unit.  At the same time, the standard Fender Reverb circuit _will_ move despite having more objects, and more types of objects, within the group I'm trying to move.

Not an emergency, I have a workaround that's good enough and I'm happy, but I saw #757 and decided to report also.  I have no real evidence that would let you reproduce the effect.  If I see anything more, I'll update this.  Thanks!

bancika, 2023-05-06:
Do you have a file that fails to move that I can use to debug?

MarsManiac, 2023-05-06:
Here you go.  I was just able to grab the entire first stage and move it.  Then I grabbed the phase inverter stage and it wouldn’t move.  Good hunting.

￼

bancika, 2023-05-07:
Didn't get the file. It doesn't work when attaching through an email. Can
you please send directly to ***@***.***?

MarsManiac, 2023-05-07:
Does this work?  If not, please be more specific about how I should send it.

MarsManiac, 2023-05-07:
If this isn’t right, please be more specific about how I should send it.

bancika, 2023-05-07:
Still don't see it. There are two ways

1) send the file as an attachment directly to me to bancikaATgmailDOTcom
2) drop the file (with the extension changed to TXT) to this github issue page and NOT to your email attachment https://github.com/bancika/diy-layout-creator/issues/774

Thanks

MarsManiac, 2023-05-07:
Weird.  Okay, I’ve dropped the TXT version into #757. Thanks for looking into this, but again, it’s not urgent to me.

bancika, 2023-05-07:
Still nothing. As long as you are replying to this email thread with at attachment it won't work.

MarsManiac, 2023-05-07:
Marshall_18W_v2.txt

MarsManiac, 2023-05-07:
Okay, please check again.  I did a straight copy and paste into the comments section of 774.  I can see the TXT file name there.

bancika, 2023-05-07:
It's only the filename, but not the actual file. Copy-paste doesn't work, the file needs to be dropped to the github page

Thank you for the patience :)

MarsManiac, 2023-05-07:
Yes, that zero-length line was the culprit.  All well now.  Thanks again.

bancika, 2023-05-17:
Fixed for 4.38

closed 2023-05-17 by bancika