bancika/diy-layout-creator #775: Can't open file -psn_0_2359872

busbyinc, 2023-05-06:
Version 4.37 running on latest version updates for Catalina.

This error only came up when opening software the first time. Subsequent opening the error did not occur. Is this something I/we should be worried about?

busbyinc, 2023-05-06:
[log-2023-05-06 01.43.44.txt](https://github.com/bancika/diy-layout-creator/files/11413408/log-2023-05-06.01.43.44.txt)

greenonline, 2023-05-18:
Yes, I saw this too, on Catalina. It appeared upon the first launch, after download. A modal dialog showed this error, but, unfortunately, I did not take a screen shot. The dialog said to refer to the log for more details. It can also be seen in the log file (as pointed out above).

bancika, 2025-02-10:
Seems to be an issue with mac. In some cases it sends process number as an argument and DIYLC expects a file name there.

This is the workaround if the issue happens

`/System/Library/Frameworks/CoreServices.framework/Versions/A/Frameworks/LaunchServices.framework/Versions/A/Support/lsregister -u /Applications/DIYLC.app`

closed 2025-02-10 by bancika