bancika/diy-layout-creator #776: zoom 25% and 33% cause the icon space on top is higher

sugizo, 2023-05-15:
tested on mac sierra 10.12.6 with latest diylc version 4.37.0
previous version is not like this (not sure what version)

25%
[image]

33%
[image]

50% or above is normal as expected
[image]

thanks and best regards

bancika, 2023-05-17:
I think it has to do with the size of the project, not the version. What happens if you double the size of the project?

sugizo, 2023-05-18:
hm not sure about the project you mentioned, but start diylc from scratch (empty new), not open *.diy file
25%
[image]

33%
[image]

50% (normal)
[image]

expected result that 25% and 33% have the same space of icon like 50%

bancika, 2026-10-19:
fixed for 6.9.0