bancika/diy-layout-creator #777: file dropdown without components

solder23, 2023-05-17:
when I open DIYLC the File dropdown menu looks like this. 
[image]

bancika, 2023-05-17:
Which distribution are you using? Universal, mac, win, flatpak?
Please upload the log file so I can invesigate.

solder23, 2023-05-17:
12:09:12,731  INFO [main] ConfigurationManager - Initializing configuration for: diylc
12:09:12,734  INFO [main] ConfigurationManager - Initializing configuration for: diylc
12:09:13,848  INFO [main] ConfigurationManager - Saving configuration
12:09:13,887  INFO [main] ConfigurationManager - Saving configuration
12:09:15,013  INFO [main] DIYLCStarter - JarLoader strategy: agent
12:09:15,014  INFO [main] DIYLCStarter - Loading languages...
12:09:15,015  INFO [main] LangUtil - Loading translation for English
12:09:15,018 DEBUG [main] LangUtil - Loading language file: /app/diylc/lang/English.txt
12:09:15,041 DEBUG [main] DIYLCStarter - Java version: 11.0.18+10 by Flathub
12:09:15,042 DEBUG [main] DIYLCStarter - Java home: /app/jre
12:09:15,042 DEBUG [main] DIYLCStarter - OS: Linux 5.15.0-71-generic
12:09:15,042  INFO [main] DIYLCStarter - Starting DIYLC with working directory /app/diylc
12:09:15,042  INFO [main] DIYLCStarter - Configuration dump start.
12:09:15,043  INFO [main] DIYLCStarter - continuousCreation = true
12:09:15,044  INFO [main] DIYLCStarter - outline = true
12:09:15,044  INFO [main] DIYLCStarter - highlightContinuityArea = false
12:09:15,044  INFO [main] DIYLCStarter - defaultBlocksImported = true
12:09:15,045  INFO [main] DIYLCStarter - Configuration dump end.
12:09:15,058  INFO [main] DIYLCStarter - JarLoader strategy: agent
12:09:15,058  INFO [main] DIYLCStarter - Loading languages...
12:09:15,059  INFO [main] LangUtil - Loading translation for English
12:09:15,062 DEBUG [main] LangUtil - Loading language file: /app/diylc/lang/English.txt
12:09:15,084 DEBUG [main] DIYLCStarter - Java version: 11.0.18+10 by Flathub
12:09:15,084 DEBUG [main] DIYLCStarter - Java home: /app/jre
12:09:15,084 DEBUG [main] DIYLCStarter - OS: Linux 5.15.0-71-generic
12:09:15,084  INFO [main] DIYLCStarter - Starting DIYLC with working directory /app/diylc
12:09:15,085  INFO [main] DIYLCStarter - Configuration dump start.
12:09:15,086  INFO [main] DIYLCStarter - continuousCreation = true
12:09:15,086  INFO [main] DIYLCStarter - outline = true
12:09:15,086  INFO [main] DIYLCStarter - highlightContinuityArea = false
12:09:15,087  INFO [main] DIYLCStarter - defaultBlocksImported = true
12:09:15,087  INFO [main] DIYLCStarter - Configuration dump end.
12:09:15,781  INFO [main] DIYLCStarter - Creating the main frame...
12:09:15,783  INFO [main] DIYLCStarter - Creating the main frame...
12:09:15,806  INFO [Thread-1] FontOptimizer - Starting font optimizer
12:09:15,813 DEBUG [Thread-1] FontOptimizer - Fonts loaded in 1ms
12:09:15,809  INFO [Thread-1] FontOptimizer - Starting font optimizer
12:09:15,826 DEBUG [Thread-1] FontOptimizer - Fonts loaded in 1ms
12:09:16,821  INFO [main] Presenter - Current DIYLC version: 4.37.0
12:09:16,919  INFO [main] Presenter - Loading component types.
12:09:16,930 DEBUG [main] Utils - Scanning jar: /app/diylc/lib/diylc-core.jar
12:09:16,970  INFO [main] Presenter - Current DIYLC version: 4.37.0
12:09:16,974 ERROR [main] Presenter - Error loading component types
java.util.zip.ZipException: invalid entry CRC (expected 0xd56ece1d but got 0x4a78c697)
	at java.base/java.util.zip.ZipInputStream.readEnd(ZipInputStream.java:410)
	at java.base/java.util.zip.ZipInputStream.read(ZipInputStream.java:199)
	at java.base/java.util.jar.JarInputStream.read(JarInputStream.java:207)
	at java.base/java.util.zip.ZipInputStream.closeEntry(ZipInputStream.java:143)
	at java.base/java.util.zip.ZipInputStream.getNextEntry(ZipInputStream.java:121)
	at java.base/java.util.jar.JarInputStream.getNextEntry(JarInputStream.java:142)
	at java.base/java.util.jar.JarInputStream.getNextJarEntry(JarInputStream.java:179)
	at org.diylc.appframework.miscutils.Utils.getFromJARFile(Utils.java:164)
	at org.diylc.appframework.miscutils.Utils.getClasses(Utils.java:124)
	at org.diylc.appframework.miscutils.Utils.getClasses(Utils.java:108)
	at org.diylc.presenter.Presenter.getComponentTypes(Unknown Source)
	at org.diylc.presenter.Presenter.<init>(Unknown Source)
	at org.diylc.swing.gui.MainFrame.<init>(Unknown Source)
	at org.diylc.DIYLCStarter.runDIYLC(Unknown Source)
	at org.diylc.DIYLCStarter.main(Unknown Source)
12:09:16,989  INFO [main] VariantManager - Checking if variants need to be updated
12:09:16,990  WARN [main] VariantManager - Could not upgrade variants for: org.diylc.components.electromechanical.RotarySwitchOpen
12:09:16,990  WARN [main] VariantManager - Could not upgrade variants for: org.diylc.components.guitar.HumbuckerPickup
12:09:16,990  WARN [main] VariantManager - Could not upgrade variants for: org.diylc.components.electromechanical.RotarySwitchSealed
12:09:16,990  WARN [main] VariantManager - Could not upgrade variants for: org.diylc.components.passive.Resistor
12:09:16,990  WARN [main] VariantManager - Could not upgrade variants for: org.diylc.components.semiconductors.DIL_IC
12:09:16,991  WARN [main] VariantManager - Could not upgrade variants for: org.diylc.components.electromechanical.CliffJack1_4
12:09:16,991  WARN [main] VariantManager - Could not upgrade variants for: org.diylc.components.guitar.SingleCoilPickup
12:09:16,991  WARN [main] VariantManager - Could not upgrade variants for: org.diylc.components.tube.TubeSocket
12:09:16,991  WARN [main] VariantManager - Could not upgrade variants for: org.diylc.components.semiconductors.BridgeRectifier
12:09:16,991  WARN [main] VariantManager - Could not upgrade variants for: org.diylc.components.electromechanical.OpenJack1_4
[... 467 lines cut ...]
[... cut ...]
12:09:22,210  INFO [main] DIYLCStarter - Injecting default properties.
12:09:22,227  INFO [SwingWorker-pool-1-thread-2] UpdateChecker - 0 updates found
12:09:22,249  INFO [main] DIYLCStarter - Main frame created.
12:09:22,249  INFO [main] DIYLCStarter - Hiding splash screen...
12:09:22,260  INFO [main] DIYLCStarter - Done hiding splash screen.
12:09:22,273  INFO [main] DIYLCStarter - Injecting default properties.
12:10:51,710  INFO [AWT-EventQueue-0] MainFrame - Closing the app
12:10:53,865  INFO [AWT-EventQueue-0] MainFrame - Closing the app

bancika, 2025-02-13:
Looks like one of the files is corrupted. Is this still an issue? I haven't seen this error and no other reports. I'm closing the issue for now

12:09:16,974 ERROR [main] Presenter - Error loading component types
java.util.zip.ZipException: invalid entry CRC (expected 0xd56ece1d but got 0x4a78c697)

closed 2025-02-13 by bancika