bancika/diy-layout-creator #780: Component request: Buzzer symbol

greenonline, 2023-05-19:
Is it possible to add a buzzer symbol? Either active, or passive, or both. Variable size.

I am currently using the "Electro-Mechanical>Fuse Holder (Panel)" symbol.

M0JXD, 2024-07-24:
Hi,

I'm working on a Buzzer schematic symbol. After re-reading this I'm unsure if your request is for a schematic symbol or the component?

I've noticed a few different buzzer types, e.g. these ones "flying lead" and mounting holes as well as PCB mount ones.

[image]
[image]

In terms of schematic symbols it seems both orientations of the semi-circle are acceptable.

[image]
[image]

For now I will do the schematic symbol, I can add the semi-circle orientation as an option/variant for each of the symbols if that is desired, although the first one seems the most common.

I'll do the PCB mount component after. Is there any desire for the version with mounting holes/flying leads? If so I can add places for flying leads to go akin to how the pickups are done.

If a datasheet of the exact part you want is available that'd be a big help! :)

greenonline, 2024-07-24:
I was after a component symbol for the stripboard layout, rather than a schematic symbol. The type of buzzer that I had in mind was the non-flying lead type, as in your second photo. 

I don't have a specific buzzer in mind, and so I don't actually know the buzzer's diameter nor the pin spacing, that is why I asked if it could be of variable diameter and pin separation.

As a general idea, like I mentioned above, I was using the *fuse holder* component symbol (for the strip board layout), which gives an approximate layout shape. You can see what I mean here: https://github.com/greenonline/AVR_High_Voltage_Fuse_Programmer_Enhanced/blob/main/README.md

M0JXD, 2024-07-26:
OK, I think I understand. I could base it on the radial electrolytic cap? That has a settable diameter and would also allow for any lead length.

It seems the active/passive tend to look the same. By default I'll make it 12mm like the ones in these links, they seem the most popular.

https://www.aliexpress.com/item/1005006213298069.html?utparam-url=scene%3Asearch%7Cquery_from%3A
https://www.aliexpress.com/item/1005006626224138.html?utparam-url=scene%3Asearch%7Cquery_from%3A

PR #827 (2024-09-20): Resistor IEC type, Buzzer Schematic Symbol and Buzzer Component
Hi Bancika,

Hope you're well, it's been a long time but I have a couple of small contributions to merge :)

First commit is for issue #798 and adds an IEC option.
Second commit adds a Buzzer Schematic Symbol. Turns out this won't close #780 sadly.

I'm hoping to do some more contributions over the weekend (famous last words...) so no rush to merge and release.

As always if there are any issues please let me know and I'll do my best to fix.
closed 2024-09-20 by bancika