bancika/diy-layout-creator #783: Creating components - walkthrough or video

greenonline, 2023-05-26:
<sup>Firstly, as I've not said it yet, this is obviously an excellent application. Many thanks.</sup>

I would really love to be able to make my own components, as I have a fair amount of free time to contribute. However, even though the [component creation guide](https://github.com/bancika/diy-layout-creator/blob/wiki/ComponentAPI.md) is probably very useful if you already _sort of_ know what to do, when starting from scratch it is a bit daunting (even for an experienced/seasoned amateur Java programmer).

A simple _walkthrough_ explaining how to create a component from scratch would be useful, and/or (and admittedly this might be more labour intensive) a YouTube video showing how to create a component. A simple component, such as a resistor.

Or even a quick walkthrough on duplicating an existing component and modifying that, if that is easier and less-labour intensive.

I think that *if* there was a more explicit guide, then there *might* be a lot more contributors to the component library.

Note: if any of these resources already exist, and I have missed them then please point me in the right direction.

---

#### A simple case study (or, case in point)

I have been looking into creating my own ZIF socket component, but not managed to get very far at all

I have spent a few hours, looking at some existing `.java` files for components (i.e. the semiconductors/IC), but it isn't immediately obvious what it going on, mostly because there aren't many comments. Now I am not saying that each and every component `.java` file should be commented to death, but having at least one such overly commented component file, as an example file, could be very useful, and serve part way towards a walkthrough.

---

#### A component editor

Clearly the best way to create a component would be to have a component editor tool, either as a standalone app or a plug-in or a built-in menu item. However, this would probably be that same amount of work as creating the initial application in the first place. Nevertheless, I did consider putting in an enhancement/feature request for such a tool, which I still could do if that is considered appropriate.

bancika, 2023-05-26:
The problem with component editor is that it would need to be really complex as some components are also very complex. They are not only for visual display but have functional aspect, like switching in conductivity. If you are considering only visuals, DIYLC is already a component editor, as you can make a building block out of other components and basic shapes. Anything more complex than that and it gets really complex, so writing code was the way to go. 

DIYLC v2 had XML files for components that allowed you to make them without coding, but functionality was limited and performance was not as good.

As for the videos/documentation...I understand where you are coming from but I have so little time for my hobbies these days, when I catch a minute to do something in DIYLC I'd rather add something that would help the end users directly, like a new component.

marko-pi, 2023-11-14:
The ability for the community to create new components relatively easily would be a big plus for the program and would take the burden of creation of the program maintainer.
I think DIYLC could be a great alternative to Fritzing, especially after Fritzing went rogue (became payable). But it should include some basic hobby components like Arduino or Raspberry Pi...

M0JXD, 2023-11-14:
I want to add my 2 pennies here as a semi-recent component contributor.

I agree a more in-depth guide/video would be useful, but my prior knowledge of Java before contributing was some playing with Processing, and prior programming experience to that being "knowing just enough C/C++ to be dangerous with an Arduino". I still don't really know Java, but the Oracle Docs and Bancika's guide got me through. All this too say that IMHO, I don't think contributing is *too* tricky, particularly with the mention of adding Arduino as a component (I've currently used building blocks) and possibly bringing Arduino hobbyists to the program.

The way the DIYLC component API works with "classes that know how to draw themselves" opens great opportunities to expand upon that. I've never tried Fritzing, but as an example, if an Arduino component was to be created it could have a "TinkerCAD circuits" style interpreter within the class. The only issue with adding Arduino/Raspberry Pi/etc. is it opens up a whole heap of modules that people will expect (i.e. Displays, Wireless Modules, sensors etc., each possibly needing it's own "pin interpreter"), which is a whole heap of work! It might be better as a plugin, because there isn't really any "simulator" support yet.

There's plenty of other cool options to explore too, take a look at this for example: https://github.com/stellartux/AudioToaster 
Essentially going from layout to (non real-time) DSP emulation.

In terms of getting more contributors, a bit of a rejig of the repository could help. Converting the Wiki branch to a GitHub wiki, and making better use of issue labels (a component request label would really help), including the "good first issue" and "help wanted" labels, could help pull in budding Java programmers looking to add to their portfolio(s). Encouraging the community to "like" and react to issues/requests does help too.

I want to get back to creating components too, but like @bancika I never seem to have the time. When I first saw this issue I intended to try making a video/improved guide the next time I contributed. I might be able to squeeze in some time around Christmas.

marko-pi, 2023-11-16:
I programm in C, C++ and Python and have used various other languages in the past, but I have never used Java. Is it possible to open the Java definitions of some examples in a normal text editor and play with them without knowing Java?

M0JXD, 2023-11-16:
You could, but I'd recommend getting Eclipse, simply because it's what @bancika has used to develop the program and you'll be able to run the code easily using the included configuration files (e.g. DIYLCStarter.launch) instead of having to setup those tasks for another development platform.

I've not really used Notepad++, but if it works for C/C++ then it should be OK for Java if you want to go that route.

In fact, the "official" Java getting started guide recommends using a box standard text editor: https://dev.java/learn/getting-started/

M0JXD, 2024-09-14:
I know this is much later than when I said, but I finally got round to making some videos regarding making components for DIYLC in Java.

Here is the link to the first one: https://www.youtube.com/watch?v=ZJ-EzqVCY24
Two more on my channel. 

They're by no means professionally made, and in all honesty I'm not much of a developer. Regardless, I hope they help!

I'm planning to make another one, but I'd appreciate feedback (if anyone watches) before doing that so I'll leave it a while before I start making another component.

closed 2025-02-13 by bancika