bancika/diy-layout-creator #785: Component request: full-sized four pin audio transformers and power transformers.

lowfrequencyoscillator, 2023-06-12:
Is there someone who knows how to program that can configure more options transformer wise. It would be awesome if I could use the software to create layouts for some mic pres.

bancika, 2023-06-12:
Existing transformers can be configured to be bigger. Try double clicking on one and see if that's sufficient. What do you need exactly?

lowfrequencyoscillator, 2023-06-13:
I figured either it wasn't part of the program yet, or I am just an idiot, and it is the latter. I did see the window for editing the component when I double clicked, but none of the parameters adjusted the type of transformer/amount of pins. What parameter exactly should I modulate? Sorry for wasting your time, I apologize for making you have to answer my dumb questions.

lowfrequencyoscillator, 2023-06-13:
> Existing transformers can be configured to be bigger. Try double clicking on one and see if that's sufficient. What do you need exactly?

I figured either it wasn't part of the program yet, or I am just an idiot, and it is the latter. I did see the window for editing the component when I double clicked, but none of the parameters adjusted the type of transformer/amount of pins. What parameter exactly should I modulate? Sorry for wasting your time, I apologize for making you have to answer my dumb questions.

closed 2025-02-13 by bancika