bancika/diy-layout-creator #787: Component Request: Terminal Block with 0.1 inch spacing

martinjking, 2023-07-02:
The existing PCB Terminal Block (in Connectivity) has 5mm pitch. I found DIYLC looking for a stripboard layout tool. It would be nice to have a terminal block to match the standard hole spacing (i.e. 2.54mm) in Veroboard/generic stripboard. The terminal blocks I use occupy approximately 3 holes (0.2 inches) of board space, front to back.

As a work-around I thought I could change the spacing of the Veroboard to 5mm so as to fit the Terminal Blocks. That does not work. The blocks snap to grid with the terminal screws over the spaces between the tracks. I want the screw heads above the copper tracks.

For now I am using lines and dots to represent the terminals. 

Thanks for making DIYLC available.
Martin.

bancika, 2023-07-02:
Hmmm, did you try moving the existing component up or down to align with
the tracks?

martinjking, 2023-07-02:
Yes indeed. It snaps (jumps) in what I assume are 5mm increments, keeping the screws above the gaps.

bancika, 2023-07-02:
You can turn off snap to grid functionality or change grid spacing to match
the increment you need

martinjking, 2023-07-02:
O.K. Snap to none applied. I think I can make it work. Newbie error. Thanks for the real-time support!

Long term, a 0.1 inch component would keep the proportions of the finished layout correct.
Thanks,
Martin.

bancika, 2023-07-03:
Not sure what the issue is. Do you have a screenshot? And a photo or tech drawing of an actual physical component that you are trying to model?

martinjking, 2023-07-03:
**To illustrate, I opened a new project.**
[image]

First image diylc_board_0.1_inch.png
Project settings, grid spacing 0.1 inch
Board properties, spacing 0.1 inch
Snap to grid on
Terminal Block (2 terminal)
This shows the size of the terminal block (5mm pitch) not matching the board (0.1 inch pitch).

**Now my work-around**
[image]

Second image diylc_board_5mm.png
Project settings, grid spacing 5mm
Board properties, spacing 5mm
Snap to grid on
Terminal Block (2 terminal)

Snap to grid working! I never saw this yesterday.

**Can I recreate the fault?**
[image]

Third image diylc_wrong_snap.png
The file I was working on yesterday
Snap to grid on
Project settings, grid spacing 0.5cm
Board properties, spacing 0.5cm
I moved terminal block that was positioned in the top left corner with snap to grid off, it still snaps with screw terminals over spaces between tracks. i.e. different behaviour to that shown above. In the file created yesterday the problem is still seen.

File saved as DIYLC_test.diy.txt, attached.
[DIYLC_test.diy.txt](https://github.com/bancika/diy-layout-creator/files/11935028/DIYLC_test.diy.txt)

**The actual device I have**
[Phoenix PCB terminal block, 2.54mm, 2 way](https://www.mouser.co.uk/ProductDetail/651-1725656) (on Mouser web site)

I hope that clarifies which components I am working with.
Regards,
Martin.

organbender, 2023-09-12:
Hi, I just installed the latest version of DIYLC (4.37). I was happy of the addition of Terminal Blocks, since I had to fake them (with 2x1 perf board, not really elegant) on previous projects. And, of course, the 2.54mm pitch is what I needed too in order to bring power to my bread board. I leave a screenshot of the context. I hope it will be available soon.
[image]

bancika, 2025-02-09:
Added for 4.40

[image]

closed 2025-02-09 by bancika