bancika/diy-layout-creator #791: Increase memory available

TheAmpBuilder, 2023-08-01:
Running on windows 11. Is there any way to increase the available memory for the app? Working on a pretty intense layout and maxing out the memory bar of 1GB.

bancika, 2023-08-01:
you can try the command below, just need to go to command prompt and
navigate to DIYLC folder. If it works, I can create a BAT file and include
it with the release. The 3rd param is the max memory limit in MB.

java -Xms512m -Xmx2048m -javaagent:lib/jar-loader.jar
-Dorg.diylc.storeDict=true -Dorg.diylc.enableTests=true
-Dorg.diylc.scriptRun=true -Ddorg.diylc.debugContinuityAreas=true
-Ddorg.diylc.debugComponentAreas=true -Ddorg.diylc.WriteStaticVariants=true
-Ddorg.diylc.WriteStaticBlocks=true --add-opens
java.base/java.util=ALL-UNNAMED --add-opens java.base/java.lang=ALL-UNNAMED
--add-opens java.base/java.text=ALL-UNNAMED --add-opens
java.desktop/java.awt=ALL-UNNAMED --add-opens
java.desktop/java.awt.font=ALL-UNNAMED --add-opens
java.desktop/java.awt.geom=ALL-UNNAMED -Dfile.encoding=Cp1252 -classpath
"C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-swing\bin;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-swing\lib\commons-compiler-3.0.6.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-swing\lib\janino-3.0.6.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-swing\lib\log4j-1.2.15.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-swing\lib\junit-4.7.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-swing\lib\jettison-1.0.1.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-swing\lib\app-framework.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-swing\lib\swing-framework.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-swing\lib\balloontip-1.2.4.1.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-swing\lib\jep-2.4.1.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-swing\lib\poi-3.5-FINAL-20090928.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-swing\lib\AppleJavaExtensions.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-swing\lib\orsonpdf-1.8.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-swing\lib\gestures-wrapper-0.2.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-swing\lib\jar-loader.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-swing\lib\svgSalamander-1.1.2.4.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-swing\lib\xstream-1.4.19.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-swing\lib\guava-31.1-jre.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-swing\lib\pdfbox-2.0.27.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-swing\lib\commons-logging-1.2.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-core\bin;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-core\lib\log4j-1.2.15.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-core\lib\junit-4.7.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-core\lib\app-framework.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-core\lib\jettison-1.0.1.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-core\lib\java-http-proxy.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-core\lib\AppleJavaExtensions.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-core\lib\xstream-1.4.19.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-core\lib\guava-31.1-jre.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-swing\lib\java-http-proxy.jar;C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc\diylc-swing\library\main.jar"
org.diylc.DIYLCStarter

bancika, 2023-08-01:
Obviously you need to replace "
C:\Users\bancika\Documents\GitHub\diy-layout-creator\diylc" with location
to DIYLC :)

thef0nz, 2024-04-07:
Idk if my issue is related to memory, but I have several projects that are unviewable or uneditable, seemingly from having too much stuff i guess. The garbage collector doesn't really help. Scrolling, editing, selecting, anything comes with a 10+ second delay.

bancika, 2024-04-08:
Can you please send some of them for me to examine? Could be something
wrong. Please zip and attach

Gebrey, 2024-07-21:
I confirm that when you have a project like a tube amp schematic, even with no AA and no HQ render the things has 10s latency. If you press more time the mouse button the layency go away for an instant. Totally a lack of memory issue.

Ive tried the command on CMD in the installation folder. No succeed. I guess if yoi have a xml line to insert in config.xml to upgrade ram avaibility. Let us know!!

bancika, 2025-03-09:
Increased to 4gb in 5.0.0

closed 2025-03-09 by bancika