bancika/diy-layout-creator #793: "Analyze Guitar Diagrams" not working as expected, Problem 1 (v4.37)

PenBach, 2023-09-03:
I really like the "Analyze Guitar Diagrams" function.
It is very useful for double-checking that a wiring diagram does what it's supposed to and there aren't any mistakes.
But I've found 2 problems.

Not sure if I'm doing something wrong, but the analysis does not seem to work correctly with the component
"DP3T (Standard 5-Position Strat)"

If we take the first row of terminals "Com  1  3  5"
and wire three pickups (HSS)
1 - PickupHB
3 - PickupSCM
5 - PickupSCN
Com - Output

then we get the expected results of a Strat 5-way lever switch.

However, if I connect the second pole's 1 & 3 connections to perform automatic coil-split of a humbucker in position 2, like this
1 - PickupHB_SplitPoint
3 - Ground

then for SW1.2, I should see PickupHB and PickupSCM at the Output, AND one of PickupHB's coils shorted to Ground
(because in Position 2 terminals 1 and 3 are connected together (and also to Com).
That is not the case. The analyze function does not find the connection between terminals 1 and 3 in Switch Position 2.
Similarly, it does not find the connection between terminals 3 and 5 in Switch Position 4.

I'll post the other problem separately as requested.

bancika, 2023-09-04:
Can you please upload DIY files with problematic layouts for both issues? Thanks

PenBach, 2023-09-04:
Sorry, it was my fault.
I've had another look, to double-check everything, and it seems I had the second pole's terminals figured incorrectly.
I could have sworn that I tried the other connection also, but...

So while the issue I reported above was down to me, it does raise the issue of identifying terminals.

The issue I've had is with the "DP3T (Standard 5-Position Strat)" lever switch and trying to figure out which terminal is which.
Apologies if I've missed something, but I can't see how to easily know which terminal corresponds to which position.
This is especially important where more than one pole is present on a switch since you need to match them up correctly.

I thought I had it figured out by wiring something up and observing the Analyze results.
Obviously I made a mistake during that process.

I tend to think of 5-way switch positions as 1=nearest bridge, 3=middle, 5=nearest neck.

The switch on my guitar is like this
5   3    1   Com
Com 1    3    5

The "DP3T (Standard 5-Position Strat)" lever switch looked physically the closest I could find to this, but I could not mirror the switch horizontally or vertically to try and match it.
Com  x   x   x
x   x    x   Com

I placed it with Angle = 270
I assumed, and thought I checked by analysing, that the switch terminals were as follows
Com  1   3    5
5   3    1   Com
I've mostly seen switches where they are "circular" with this kind of pattern, but there are variations.

But, double-checking this again by analysing, it looks like they are actually (Angle = 270)
Com  1   3    5
1   3    5   Com
And at Angle = 90, it's
Com  5   3    1
5   3    1   Com

So...
Is there a way of telling which terminal on the lever switch corresponds to which position when drawing?
The only way I could find was to wire it and analyse.
And obviously that process is prone to error, as demonstrated here.

Could I request something similar to the "Mark Common Lugs" checkbox is introduced for the terminals?
How about a checkbox for "Label All Terminals"?
That way, it can be switched on when wiring, and off if too intrusive.
Or, identifying terminals in the Status bar when mouse over or connecting wires would be helpful.
And a switch that matches my guitar's switch would also be good, since a diagram using the provided switch would be confusing/misleading.
Should I raise a separate Feature request for this?

As always, your work is appreciated. Thanks.

PenBach, 2023-09-13:
What "can't be"?
I can try and give more detail if necessary.

bancika, 2023-09-13:
Because angle of rotation is not considered for anything except for
rendering. I checked the code.
A DIY file with both versions of the switch would be great.

PenBach, 2023-09-14:
I think we're getting a little side-tracked here.
I've attached some files for you to look at (with ".txt" added to upload).

Again, the main problem for me with this switch is that it's difficult to know which terminals are which.
While that's open to interpretation on a 1-pole switch, it is essential to match up corresponding terminals on switches with more than 1 pole.
So could you please consider adding something like I asked for in my second comment above?
Maybe just add a "Mark Lug 1" checkbox similar to the "Mark Common Lugs" checkbox, in a different colour, since you already have that functionality in the code. The more information the better.
[Test_Strat5way3PDT_793_270degP2.diy.txt](https://github.com/bancika/diy-layout-creator/files/12608604/Test_Strat5way3PDT_793_270degP2.diy.txt)
[Test_Strat5way3PDT_793_270degP1.diy.txt](https://github.com/bancika/diy-layout-creator/files/12608605/Test_Strat5way3PDT_793_270degP1.diy.txt)
[Test_Strat5way3PDT_793_90degP1.diy.txt](https://github.com/bancika/diy-layout-creator/files/12608606/Test_Strat5way3PDT_793_90degP1.diy.txt)
[Test_Strat5way3PDT_793_90degP2.diy.txt](https://github.com/bancika/diy-layout-creator/files/12608607/Test_Strat5way3PDT_793_90degP2.diy.txt)

astromiester1, 2024-03-18:
Thats a great idea... Allow a user checkbox to see terminal numbers and letter assignments.... Hope he does this...

bancika, 2025-02-13:
I added a ticket for the feature that will solve this issue in a better way https://github.com/bancika/diy-layout-creator/issues/873

Closing this one

closed 2025-02-13 by bancika