bancika/diy-layout-creator #794: "Analyze Guitar Diagrams" not working as expected, Problem 2 (v4.37)

PenBach, 2023-09-03:
Analyze Guitar Diagrams often gives TreeException and does not give any Analysis results.

I often get the error MessageBox
"org.diylc.netlist.TreeException: Could not find anything connected to a jack tip terminal"

Im guessing it may be caused when for example a lever switch does not have an "Output" for all of its positions, even though that is a valid setup.

Using the "DP3T (Standard 5-Position Strat)",
If we take the row of terminals "Com  1  3  5"
and wire two pickups
1 - PickupHB
3 - PickupSCM
5 - No connection
Com - Output

Gives the error/exception.

This is from the log (v4.37)
Caused by: org.diylc.netlist.TreeException: Could not find anything connected to a jack tip terminal.
	at org.diylc.netlist.GuitarDiagramAnalyzer.constructTree(GuitarDiagramAnalyzer.java:364)
	at org.diylc.netlist.GuitarDiagramAnalyzer.summarize(GuitarDiagramAnalyzer.java:101)
	at org.diylc.netlist.NetlistAnalyzer.summarize(Unknown Source)
	at org.diylc.swing.ActionFactory$SummarizeNetlistAction$1.doInBackground(Unknown Source)
	at org.diylc.swing.ActionFactory$SummarizeNetlistAction$1.doInBackground(Unknown Source)
	at org.diylc.swing.gui.MainFrame$6.doInBackground(Unknown Source)
	at javax.swing.SwingWorker$1.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at javax.swing.SwingWorker.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)

PenBach, 2023-09-04:
I've attached a diagram as requested.
I had to tag on ".txt" for GitHub to accept it. Just remove the extra ".txt" extension.

[Test_Strat5way3PDT_794.diy.txt](https://github.com/bancika/diy-layout-creator/files/12515027/Test_Strat5way3PDT_794.diy.txt)

DeltaPotential, 2023-10-23:
I finished mocking up the wiring of a new, to me, guitar and I get the same error; log [below] is slightly different. I also tried to use the analyze feature on @PenBach's file and I'm returned the same error.
 
OS: MacOS v12.6.9
diylc v4.37.0

> 02:06:47,188  INFO [SwingWorker-pool-1-thread-4] ConfigurationManager - Saving configuration
02:06:51,885 ERROR [AWT-EventQueue-0] MainFrame - Background task execution failed
java.util.concurrent.ExecutionException: org.diylc.netlist.TreeException: Could not find anything connected to a jack tip terminal.
	at java.util.concurrent.FutureTask.report(FutureTask.java:122)
	at java.util.concurrent.FutureTask.get(FutureTask.java:192)
	at javax.swing.SwingWorker.get(SwingWorker.java:602)
	at org.diylc.swing.gui.MainFrame$6.done(Unknown Source)
	at javax.swing.SwingWorker$5.run(SwingWorker.java:737)
	at javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.run(SwingWorker.java:832)
	at sun.swing.AccumulativeRunnable.run(AccumulativeRunnable.java:112)
	at javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.actionPerformed(SwingWorker.java:842)
	at javax.swing.Timer.fireActionPerformed(Timer.java:313)
	at javax.swing.Timer$DoPostEvent.run(Timer.java:245)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
**Caused by**: org.diylc.netlist.TreeException: Could not find anything connected to a jack tip terminal.
	at org.diylc.netlist.GuitarDiagramAnalyzer.constructTree(GuitarDiagramAnalyzer.java:364)
	at org.diylc.netlist.GuitarDiagramAnalyzer.summarize(GuitarDiagramAnalyzer.java:101)
	at org.diylc.netlist.NetlistAnalyzer.summarize(Unknown Source)
	at org.diylc.swing.ActionFactory$SummarizeNetlistAction$1.doInBackground(Unknown Source)
	at org.diylc.swing.ActionFactory$SummarizeNetlistAction$1.doInBackground(Unknown Source)
	at org.diylc.swing.gui.MainFrame$6.doInBackground(Unknown Source)
	at javax.swing.SwingWorker$1.call(SwingWorker.java:295)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at javax.swing.SwingWorker.run(SwingWorker.java:334)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:750)

bancika, 2023-10-23:
please upload your DIY file so I can replicate the issue locally.

DeltaPotential, 2023-10-24:
@bancika, thank you for kindly offering to take a look. I've attached the diagram.

This is my first foray into guitar wiring and switches, so it is possible that I have made a mistake. However, I was able to reproduce the issue with @PenBach's file too.

I noticed the ground connections don't "stick" to the pots or metal plate of the 5-way switch in the diagram. Is that expected?

[Ibanez WIring - As Bought.diy.txt](https://github.com/bancika/diy-layout-creator/files/13092309/Ibanez.WIring.-.As.Bought.diy.txt)

DeltaPotential, 2023-11-17:
I'd greatly appreciate if anyone could take a look and let me know if I made a mistake.

FlounderJr, 2024-04-18:
I'm having the same issue with the error:
"org.diylc.netlist.TreeException: Could not find anything connected to a jack tip terminal"

Additionally, I have this question as well:
"I noticed the ground connections don't "stick" to the pots or metal plate of the 5-way switch in the diagram."

Has anyway resolved these?

juanpanchoec, 2024-12-02:
I have the same issue. It works fine if a pickup is wired to a jack, but adding even the simplest mini-switch (SPST) in the middle of either wire the exact same error (org.diylc.netlist.TreeException: Could not find anything connected to a jack tip terminal) pops up when trying to analyze.
If the switch is On/On, then adding a jumper/wire in the position that would close the circuit if selected makes the analyzer work, simulating both positions. As OP wrote: "Im guessing it may be caused when for example a lever switch does not have an "Output" for all of its positions, even though that is a valid setup."

bancika, 2025-02-13:
The next version (4.40) will be able to work even when some of the switch combinations produce no sound.

@FlounderJr please file a separate request for the ground connection not sticking to the pots. It's by design, only control points can be sticky, not the entire areas. But maybe there's a possibility to enhance in the future. It's best to have a separate ticket as it's impossible to track multiple requests per ticket.

Closing this one as it's done

closed 2025-02-13 by bancika